balena-io/etcher #470: support rich image extensions, starting with simple metadata

alexandrosm, 2016-06-11:
With the support etcher has for compressed images, we have the opportunity to ask image creators to add an *optional* metadata folder, which will hold additional info that etcher can use to improve the user experience. 

Straightforward information included in the folder could be: 
- a user-readable name for the image (e.g. "Raspbian Jessie - May 2016" instead of "2016-05-27-raspbian-jessie.zip")
- a URL with to further information about the release (e.g. http://downloads.raspberrypi.org/raspbian/release_notes.txt)
- a thumbnail to replace the "white disc" image above the image name

More advanced ideas would be things like
- a bmap file to speed up writing times
- how to configure the image (e.g. wifi configuration etc) that Etcher could automate
- validating that the image is what it says it is, and that it is the latest version (requires online integration?)

This issue suggests adding support for the first three items, opening the path for the next ones in the future.

The existence of each 

A compliant compressed image file, when expanded, should also contain a directory, named "_info", which will contain assets and metadata. Another naming idea is to call the directory "_etch" to clarify that this metadata is intended for etcher. It's unclear which choice is cleaner from a community perspective.

Due to the wide adoption by developers, the JSON format is probably an ideal choice for the manifest file. The file could be called "manifest.json". Example contents could be:

```
{
    full_name: "Raspbian Jessie - May 2016",
    further_info: {
        url: "http://downloads.raspberrypi.org/raspbian/release_notes.txt",
        text: "release notes"
    },
    logo: "<filename in folder> or <URL>"
}
```

All these attributes refer to the image that is to be written, so we avoid using the word "image" in the field names. It is implied. In a world where multiple images were included in an archive (not a feature we want to support right now, just an example), the file could look like so:

```
{
    images: [{...}, {...}, ...]
}
```

In the above each instance of {...} would be an instance of the manifest above. I write this second example to clarify the point of view from which the modeling for the manifest has been done. It is intended to list attributes about a specific disk image.

Each attribute (full_name, further_info, logo) should be useful on its own and not depend on the existence of the others. The elements in further_info of course can be dependent on each other, which is why they have been grouped, as they reflect a single unit.

The specification on the logo should include the size it will capture on the ui. We could limit the supported formats but "whatever chrome is ok with" is probably good enough?

Thoughts?

jviotti, 2016-06-11:
Looks very good, and doing the first steps is luckily something that can be accomplished without much effort (given we already have all the building blocks in place).

Some comments:

> a URL with to further information about the release

How do you envision this? Should we just link to it from somewhere?

> A compliant compressed image file, when expanded, should also contain a directory, named "_info", which will contain assets and metadata. Another naming idea is to call the directory "_etch" to [...]

I vote for a generic name. Imagine if other tools (maybe not strictly image writers, but complementary software) also implement support for this, and we can push it as an "standard" way to describe OS images.

```json
{
    full_name: "Raspbian Jessie - May 2016",
    further_info: {
        url: "http://downloads.raspberrypi.org/raspbian/release_notes.txt",
        text: "release notes"
    },
    logo: "<filename in folder> or <URL>"
}
```

We should strive to make the property names as similar to Node's `package.json` as possible, for consistency. We also have to be very careful about this, since if this idea gets widespread adoption, it will be very hard for us to introduce breaking changes. We should take proper planning time before shipping anything.

```json
{
    images: [{...}, {...}, ...]
}
```

JSON supports top level arrays. This can be done like this:

```json
[
    {...},
    {...},
    {...}
]
```

> The specification on the logo should include the size it will capture on the ui. We could limit the supported formats but "whatever chrome is ok with" is probably good enough?

Retina support is something to consider here, therefore we should limit the options to image formats that can contain multi-layer images, supporting various resolutions (which I wouldn't recommend, since the process is usually not very straightforward and some people will ignore this and will make Etcher look ugly on certain devices), or an SVG.

alexandrosm, 2016-06-11:
On the envisioning part, I see this looking something like:

[logo]
[name] (instead of filename)
[more info link] (under name)

we may have to work with Taahir on the details of how this looks.

alexandrosm, 2016-06-11:
I'm with you on the generic name.

On the package.json consistency, I'd like to see what that looks like, as it's a completely different thing being described here. Can you give an example?

alexandrosm, 2016-06-11:
The idea to consider retina support is a good one. Let's limit to that, then :). We should also guide image creators that they should probably use transparency to make it look good, and their logo should look good on both dark and light backgrounds, as we can't guarantee that etcher's background won't change in the future.

alexandrosm, 2016-06-11:
If we manage to build an alpha version of this, we should leave it undocumented to begin with, then show how this works to some people who create images and get their feedback.

jviotti, 2016-06-11:
I agree with every comment. 

> On the package.json consistency, I'd like to see what that looks like, as it's a completely different thing being described here. Can you give an example?

Maybe something like this:

```json
{
    "name": "image-code-name",
    "description": "Raspbian Jessie - May 2016",
    "version": "1.2.0",
    "homepage": "http://raspberrypi.org/raspbian",
    "logo: "rpi.svg"
}
```

Notice my `logo` property doesn't allow a URL. This is because:

- We don't want to worry about online status.
- The link might broke someday. If the image is inside the archive, everything is self-contained, and will work forever.

Regarding the release notes, I think users would prefer to be taken to the image website, and go to any subsection they are interested in (e.g: release notes) by themselves.

alexandrosm, 2016-06-11:
the package.json equivalence looks forced to me. That format was defined for something very different (a node.js package) and here we're trying to describe something else (a disk image, likely of an OS or application). I agree we should be careful not to choose names we will change in the future, but unfortunately, I don't think there exists a standard that can help us for this.

On the release notes, my reasoning is this: Users may want to easily find out what is being written to their drive. They can probably find the project website, but the release notes for the specific version may be harder to find. 

I could see a use for a "version" field, though.

Now that I think of it, if we had a <name> <version> line under the image (replacing the filename), both of these could be links, with the <name> linking to the project website, and the <version> linking to the release logs. What do you think?

I am ok with dropping the URI for the logo. I had some exotic use cases in mind, but it's simpler to leave it out.

jviotti, 2016-06-12:
> the package.json equivalence looks forced to me. That format was defined for something very different (a node.js package) and here we're trying to describe something else (a disk image, likely of an [...]

Are you thinking about a specific property that looks forced? Most of the properties introduced by `package.json` are very generic, and should apply without issues to any other resources, like `name`, `version`, etc. I do feel `description` is a bit forced. Maybe `display_name` is better?

> On the release notes, my reasoning is this: Users may want to easily find out what is being written to their drive. They can probably find the project website, but the release notes for the specific version may be harder to find.

Yeah, I guess we should offer both. I foresee users trying to link to many things from Etcher, so we should provide a sensible list of things that can be linked to that cover the majority of OS images publishers.

> Now that I think of it, if we had a line under the image (replacing the filename), both of these could be links, with the linking to the project website, and the linking to the release logs. What do you think?

Sounds good to me.

> I am ok with dropping the URI for the logo. I had some exotic use cases in mind, but it's simpler to leave it out.

Do you mind expanding on your exotic use cases ideas, as a way to document them there for the future?

alexandrosm, 2016-06-22:
The exotic use cases I was thinking about were things like allowing the image maintainers to gather some analytics about use of their images by adding querystring parameters to their images. So a url such as someboard.org/logos/logo_small.png?img_version=1.1.1&edition=industrial or whatever. But it's a bit too exotic, and I'm not sure how users will like a feature like that.

alexandrosm, 2016-06-22:
On the package.json, "description" feels forced for sure. I think in order to avoid changing the standard in the future, we should start with something simple that fits our use case exactly, since there is no real "ecosystem" benefit or tool re-use from standardising on package.json

jviotti, 2016-07-11:
I've just PRed the first step towards this feature, which support for the `manifest.json` file (specifically the name property). Now that we have the machinery in place, adding the URL is trivial. 

Can we get a mock from @taahirisaacs as a way to start discussion on how/where to display the readable image name and URL?

jviotti, 2016-07-11:
Support for `manifest.json` and an SVG logo landed in `etcher-image-stream` v2.4.0.

jviotti, 2016-07-26:
I've just PRed (graphical) support for this at https://github.com/resin-io/etcher/issues/597. The design has not been fully discussed, but this PR is a great way to get the ball rolling. I'll close this issue with that PR, and we can open new ones to improve styling, etc.

closed 2016-07-26 by jviotti