balena-io/etcher #480: Provide a way for users to bypass the Etcher system drive protection layer

jviotti, 2016-06-13:
There might be advanced users that really know what they're doing.

We could either:

- Document the use of an environment variable, like `ETCHER_ADVANCED_MODE=1`. Advanced users will feel confortable doing so, and the feature will be nicely isolated from newbies.

- Provide a checkbox in the drive selector modal saying something like "Advanced usage" with a big warning that will make system drives appear on the modal.

msupon, 2016-06-14:
I like the env var option, and I appreciate the distro without the check in it.
[image]

booted off the drive and it worked perfectly.

jviotti, 2016-06-14:
Sounds good. Possible names for the environment variable:

- `ETCHER_ADVANCED_MODE`
- `ETCHER_UNSAFE_MODE`
- `ETCHER_RUN_UNSAFE`

I personally like the second one.

alexandrosm, 2016-06-22:
@jviotti how about an option in the settings?

jviotti, 2016-06-22:
@alexandrosm Yeah, I think that'd work even better. Propagating env vars from the main process to the renderer process is a pain, so I'm all for a more simplistic option.

alexandrosm, 2016-06-22:
maybe with a red "dangerous" label next to it, and a warning modal when the user wants to enable it?

jviotti, 2016-06-22:
I personally think the dangerous label is enough, and a warning modal is just annoying, but maybe that's just me. @taahirisaacs?

alexandrosm, 2016-06-22:
they'll only see it once anyway, not a big deal, and want to make sure
someone doesn't blow out their hard drive and blame etcher

jviotti, 2016-07-01:
[image]

closed 2016-07-19 by jviotti

jmsaltzman, 2019-02-22:
I found this thread looking for details on this setting, and while "Allow writing system drive [DANGEROUS]" or other text for this setting would be more correct, this text, as-is, is too hilarious to ever change!!

"Unsafe mode" is mysterious, but the "Dangerous" makes it delicious as well!

It's like a red button with a label next to it saying "DO NOT PRESS THIS BUTTON."

lurch, 2019-02-22:
https://github.com/balena-io/etcher/blob/master/lib/gui/app/pages/settings/templates/settings.tpl.html#L63
:slightly_smiling_face: :fire: