balena-io/etcher #481: Etcher won't allow writing via USB SD-Card adapter

tobyjaffey, 2016-06-14:
Using an external USB SD-Card adapter, OSX disk utility and command line tools can work with SD-Cards. When I use etcher, the device is shown with strikethrough text and the button is disabled.

Is it possible to use Etcher with an external USB adapter?

I am using MacOSX El Capitan + Integral external USB adapter (vid=0x0bda, pid=0x0119)

jviotti, 2016-06-14:
Hey there,

A strikethrough device currently means that the device is not large enough to be flashed with the image you selected. Can you check the drive capacity to confirm this?

We know this is very confusing, and a nice red label saying that the drive is too small will be shown in the next release.

tobyjaffey, 2016-06-14:
You're right, my 4GB SD-Card is actually just under. Thanks.

closed 2016-06-14 by tobyjaffey