balena-io/etcher #482: Etcher installation won't work with custom path /tmp folder

didli, 2016-06-15:
Not sure if it's an issue with AppImage or Etcher, but I couldn't install and run Etcher (stuck in a loop) until I unmount my tmp (bind on another partition than /) folder.

> mkdir: impossible de créer le répertoire « /tmp/.mount_H5tEBw »: Le fichier existe
sh: ligne 1: /mnt/cache/.tmp/.mount_H5tEBw-elevated: Aucun fichier ou dossier de ce type
sh: ligne 2: /mnt/cache/.tmp/.mount_H5tEBw-elevated: Aucun fichier ou dossier de ce type
sh: ligne 3: /mnt/cache/.tmp/.mount_H5tEBw-elevated/usr/bin/etcher: Aucun fichier ou dossier de ce type
umount: /tmp/.mount_H5tEBw : cible occupée
       (Dans certains cas, des renseignements sur les processus utilisant
        le périphérique sont accessibles avec lsof(8) ou fuser(1).)
sh: ligne 4: /mnt/cache/.tmp/.mount_H5tEBw-elevated: Aucun fichier ou dossier de ce type

jviotti, 2016-06-15:
Hi @didli ,

Thanks for reporting this. I believe its related to the following issue: https://github.com/resin-io/etcher/issues/459. We are performing some nasty hacks to workaround a FUSE limitation in AppImages, which seems to be causing this.

Etcher gets stuck in the loop you mention above if the application is not existed gracefully (e.g: by force-killing the process with Ctrl-C or something like that).

jviotti, 2016-06-15:
I'll close this one, but will keep it referenced from the other issue!

didli, 2016-06-15:
I've seen this issue. I'm not entirely sure it is related, since i didn't forced-closed anything and removing my tmp folder fixed the problem. I'm able to reproduce the error everytime : 
- When i'm using a normal tmp path : no problem.
- When I mount the customized tmp path in my fstab : Etcher fails everytime. Thing is I'm thinking now it's not related to Etcher at all... Sorry for that - -'

jviotti, 2016-06-15:
Hi @didli ,

> When I mount the customized tmp path in my fstab : Etcher fails everytime. Thing is I'm thinking now it's not related to Etcher at all... Sorry for that - -'

Oh, that's very interesting. Do you mind sharing details about such customised tmp path? Are you using a custom FS for it, or any special mount option?

I'd be cool to know to either account for it on Etcher, or deal with it upstream in AppImages.

didli, 2016-06-15:
Just an old line in fstab to avoid using my SSD and bind tmp, log and cache to another disk : 
/mnt/cache/.tmp  /tmp  bind	defaults,bind,noatime,mode=1777	0	0
Maybe the dot .tmp is not a great idea ^^
I've just tried others appimages (Brackets ...) which seems to works perfectly fine even with my strange tmp path. Note that it's not really going to prevent me from using Etcher (by the way, thank you for it, it's flawless, and very reliable), i will just make a script that disables my tmp path everytime I need Etcher and make a launcher from this ^^.

jviotti, 2016-06-15:
Thanks for the information and for the nice words! I'll rename this issue, keep it open and hopefully try to reproduce and fix very soon!

Thank you very much again for the support!

jviotti, 2016-06-15:
@didli Do you mind showing me the full output of the `mount` command when you attempt to run Etcher with your custom `/tmp`?

jviotti, 2016-06-15:
In order to offer some context:

Etcher needs `root` privileges in order to write to devices directly. AppImages are mounted with FUSE, which only allows the user that mounts the image to execute anything inside. So if I execute the AppImage as `jviotti`, and then attempt to call a binary inside as `root`, FUSE will throw `EPERM`.

As a way to workaround this, we do the following:

- Mount the AppImage normally, as the current user.
- If the user is `root`, we carry on, and the app executes correctly.
- If the user is not `root`, we show an elevation dialog.
- If the user passes the password challenge imposed by the elevation dialog, the original AppImage is re-mounted again as `root`, and re-executed as that user.

The logic that performs this re-mounting stuff lives in the following Bash script: https://github.com/resin-io/etcher/blob/master/scripts/elevate-linux.sh

didli, 2016-06-15:
- If the user passes the password challenge imposed by the elevation dialog, the original AppImage is re-mounted again as root
And i suppose there lies my problem, because everytime the only error that the terminal is throwing at me is that it can't mkdir because the folder already exists, and it can't mount either the elevated folder because it can't find it (which I don't understand at all). 
Unfortunately, the only output I can give you is the one I posted first. Can't get my hands on anything else (at the moment). Is there a flag I can try to give you more information ?

jviotti, 2016-06-15:
Hi @didli ,

I believe I found the issue (and have at least a workaround). The problem seems to lie in the line that attempts to find a good mount-point for the re-mounted image:

```sh
// See https://github.com/resin-io/etcher/blob/master/scripts/elevate-linux.sh#L42
mountpoint=$(mount | grep $(basename $APPIMAGE) | awk '{ print $3 }')-elevated
```

`$(basename $APPIMAGE)` usually evaluates to `Etcher-linux-x64.AppImage`. If for some reason the application did not succeed at un-mounting the previous re-mounted mount point the last time it ran, the `mount` command will have many entries containing the string `Etcher-linux-x64.AppImage`, therefore `grep` will return a multi-line string.

Awk will then proceed to print the third column of *every* line in the multi-line string, so the final value of `mountpoint` will be something like:

```
/tmp/.mountXXXX
/tmp/.mountYYYY-elevated
```

When we finally call `mkdir -p $mountpoint`, this will be evaluated to:

```
mkdir -p /tmp/.mountXXXX
/tmp/.mountYYYY-elevated
```

So `/tmp/.mountXXXX` will get created, and the system will try to erroneously run `/tmp/.mountYYYY-elevated` as a separate command, causing very weird errors.

Since the third column of the `mount` output may not be a path to a directory, the error messages might get even more cryptic (I was able to get very weird ones right now).

I'll build a new AppImage containing a workaround (that simply makes sure the mount-point gets assigned correctly), so you can confirm it works for you.

This of course doesn't tackle the root cause, which is Etcher not un-mounting the elevated directory in some cases, which is something still needs some investigation.

I'll ping you once I upload the build mentioned above.

jviotti, 2016-06-15:
Turns out FUSE takes care of un-mounting the image as soon as its process exits. I doubt we can as powerful as that here :/

didli, 2016-06-16:
This is it, you nailed it. I really appreciate your work on this and I must say, really glad about your feedback here. Usually, I'm just given a vague answer stating that the issue is resolved but you go all the way explaining things, and for this : **thank you**  !
Now, please stop working on this ^^ ! Especially if fuse or whatever is getting in your way : I can deal with the tmp location just for Etcher so you can concentrate on making Etcher even better, solving others users issues. And again, thank you for your time. Much appreciated.

jviotti, 2016-06-16:
Hi @didli ,

> This is it, you nailed it. I really appreciate your work on this and I must say, really glad about your feedback here. Usually, I'm just given a vague answer stating that the issue is resolved but you go all the way explaining things, and for this : thank you !

Super happy to hear that, and thank you very much for your very kind words. We're trying very hard to lower the barriers of contributions (bug reports, patches, docs, etc) for this project so everyone can have a warm friendly welcome to the project.

Thank you very much for taking the time to report an issue, and sticking around willing to try things we throw at you, that's extremely valuable for us!

BTW, you might want to join our Gitter channel (https://gitter.im/resin-io/chat) to say hi!

> Now, please stop working on this ^^ ! Especially if fuse or whatever is getting in your way : I can deal with the tmp location just for Etcher so you can concentrate on making Etcher even better, solving others users issues. And again, thank you for your time. Much appreciated.

Yeah, I'm currently working on a work-in-progress pull request (https://github.com/resin-io/etcher/pull/423) to only elevate the writer process instead of the whole application, which should at least mitigate this issue. In any case, I'll get in touch with the FUSE team to see if they have any feedback about this.

I'll make a PR with the patch shared above and it will be included in the next release. For now you can just use the custom build I sent you to avoid having to mess with your `fstab`.

jviotti, 2016-06-16:
@didli ,

Ah lol, I never actually sent you the build, sorry for that :D

Here it is: https://www.dropbox.com/s/cqaefajhyp1aawr/Etcher-linux-x64.AppImage?dl=0

closed 2016-06-16 by jviotti