balena-io/etcher #491: Support for Ostro Project .dsk files

rexstjohn, 2016-06-17:
Hello, I would like to request support for the .dsk files created by Ostro Project Linux (https://github.com/ostroproject/ostro-os/releases/).

rexstjohn, 2016-06-17:
More details: https://ostroproject.org/documentation/howtos/booting-and-installation.html

jviotti, 2016-06-17:
Hi @rexstjohn ,

Sounds very interesting. So `*.dsk` files can be written directly to the drive (after uncompressing, if necessary)? Assuming so from the following example:

```sh
$ sudo umount /dev/sdb*
$ xzcat <ostro-os-image.dsk.xz> | sudo dd of=/dev/sdb bs=512k
$ sync
```

`bmaptool` is something we have plans to work on very soon, however it also means we have to port from Python as a NodeJS write stream, otherwise it might get tricky to add support to it in a cross platform way.

If `*.dsk` files can be written normally, I can add support very soon, otherwise we'll have to wait until we sort out the `bmaptool` re-implementation.

closed 2016-06-22 by jviotti

jviotti, 2016-06-28:
Hi @rexstjohn ,

I'm happy to announce DSK support landed in v1.0.0-beta.10, which you can get from the website: http://www.etcher.io