balena-io/etcher #492: Cannot select an ISO [1.0.0-beta.8]

cassidyjames, 2016-06-18:
Selecting an ISO, whether via drag-and-drop or via the file dialog, is not working in Linux with 1.0.0-beta.8. It worked with the previous version.

Etcher never proceesds to the next step. Instead, it stays on the first step as if I had not selected an ISO.

oliver-ruehl, 2016-06-18:
Convfirmed, happens to me on Mac OS X El-Capitan 64-Bit

jviotti, 2016-06-18:
Hey there,

Thanks for reporting guys. It's strange, I can select an ISO correctly on both Ubuntu 16.04 and OS 10.11.4, at least a CoreOS one I happen to have here.

Can link me to the specific ISO you're trying to select? Also, what's the file mime type according to the `file` command?

oliver-ruehl, 2016-06-18:
`elementaryos-0.3.2-stable-i386.20151209.iso: ISO 9660 CD-ROM filesystem data 'elementary OS                  ' (bootable)`
Download location: https://elementary.io/

jviotti, 2016-06-18:
The only thing I think could be failed my is the following function: https://github.com/resin-io/etcher/blob/master/lib/gui/models/supported-formats.js#L109, which we use as a simple mechanism to detect if the image is valid or not, just by looking at the file name.

Can you try opening DevTools by pressing Cmd+Alt+I or Ctrl+Alt+I and see if you get a log message saying that the Image you tried to select is invalid? If so, please share the file name and I'll fix, and you can rename the file to something simpler that doesn't confuse Etcher as a workaround.

oliver-ruehl, 2016-06-18:
`Sat Jun 18 2016 15:54:58 GMT+0200 (CEST) Invalid image ({"path":"/Users/murmelmurmel/Downloads/elementaryos-0.3.2-stable-i386.20151209.iso","size":1126170624})`

travm, 2016-06-18:
I can confirm that I am getting the same issue. I'm also trying to use it with Elementary OS - so the error is the exact same.

tinywook, 2016-06-18:
I had the issue and was able to fix it by changing the filename from xxx.xxx.xxx.img to xxx.img

jviotti, 2016-06-18:
Thank you very much everyone. I'm glad we at least have a workaround. I'll push a hot fix very soon, and sorry for the inconveniences!

jviotti, 2016-06-18:
Elementary OS is adding a timestamp as if it was an extension: `.20151209.iso` which is confusing Etcher.

I've locked the Etcher website to `v1.0.0-beta.7` until the fix is released, so no more new users encounter this. I'll push a fix and publish a new patched version, sorry again for the troubles!

albinekb, 2016-06-19:
Same issue, worked after renaming to `xxx.iso` instead of `xxx.xxx.xxx.iso`

jviotti, 2016-06-20:
Hi everyone,

Sorry for taking so long to merge this. I decided to take some time to really ensure this is the first and last time a similar issue happens to our users by increasing our test coverage on this area, and working on a new algorithm to detect if an image is supported that deals with these corner cases.

I'm publishing v1.0.0-beta.9 as I write, and will ping you here in the same thread once its live.

Apologies, and have a great day!

closed 2016-06-20 by jviotti

jviotti, 2016-06-21:
I've just published v1.0.0-beta.9 with the fix, which is now available in the website.