balena-io/etcher #496: Problems with dots in MacOSX 10.10.5

rafaeltovar, 2016-06-20:
Hi, Etcher don't open image files when have dots in the name. For example 'elementary-3.0.8.iso'. In MacOSX 10.10.5.

jviotti, 2016-06-20:
Hi @rafaeltovar ,

This is a duplicate of  https://github.com/resin-io/etcher/issues/492, which has just been closed. I'm uploading a new version with a fix as I write, stay tuned!

closed 2016-06-20 by jviotti

rafaeltovar, 2016-06-20:
Oh! Sorry! :(

jviotti, 2016-06-21:
No need to feel sorry! Thanks for reporting this, and forgive us for introducing this regression that annoyed so many Elementary OS users.

You went all the way to discover the workaround I shared in the other issue by yourself, which is super awesome!

jviotti, 2016-06-21:
I've published v1.0.0-beta.9 with the fix, which you can get from the website!