balena-io/etcher #509: Wayland support?

GrzegorzBB, 2016-06-22:
Wayland support?

jviotti, 2016-06-22:
Hi @GrzegorzBB ,

I've never ran Etcher on Wayland, but I suspect it should just work. Do you see any error?

GrzegorzBB, 2016-06-22:
Not much:

```
/tmp/.mount_42P7kG/usr/bin/elevate.wrapper
/usr/bin/pkexec
No protocol specified
```
My display env:

```
WAYLAND_DISPLAY=wayland-0
DISPLAY=:0
```

jviotti, 2016-06-22:
I see, I'll give this a go myself. In the mid-time, what happens if you run the appImage as `root`?

GrzegorzBB, 2016-06-22:
Almost the same:

```
[root@lenovog570 Pobrane]# ./Etcher-linux-x64.AppImage 
/tmp/.mount_GyOEd1/usr/bin/elevate.wrapper
```

jviotti, 2016-06-22:
Weird, I'll test it myself and report back as soon as possible.

GrzegorzBB, 2016-06-22:
Thank you Juan ;).

jviotti, 2016-07-07:
Hi @GrzegorzBB ,

Looks like setting `GDK_BACKEND=x11` (see https://github.com/electron/electron/issues/2927#issuecomment-227598167) does the trick. Can you confirm that works for you as well?

If so, I'll add this environment variable to the AppImage itself, so it gets set automatically.

GrzegorzBB, 2016-07-08:
Sorry Juan, no change for me.
```
[Grzegorz@lenovog570 Pobrane]$ env |grep GDK
GDK_BACKEND=x11
```
If you want any additional info just tell :).

jviotti, 2016-07-08:
@GrzegorzBB Too bad! What distribution are you running? Is there a way I can get a system similar to yours? I've tested this under Weston on RebeccaBlackOS and it worked just fine.

GrzegorzBB, 2016-07-10:
My distribution is Fedora 24 Workstation, i use wayland session. Every Fedora 24 will be very similar to my system.

jviotti, 2016-08-04:
Sorry for the huge delay on this ticket. Its been pushed aside by other priorities, but I'm getting back to it soon. Some thoughts:

- Looks like GTK-2 doesn't have proper Wayland support. There seems to be a way to create a custom Electron build on top of GTK-3, which should solve the issue. See https://github.com/nikolowry/electron-gtk3.

- The application *should* be running fine with GTK-2 thanks to [XWayland](https://wayland.freedesktop.org/xserver.html) in any case, since it provides backwards compatibility for *any* X clients. As far as I researched, Fedora should handle XWayland stuff out of the box when running X applications. What I believe could be happening (and this might be related to the reason why my environment variable suggestion above didn't work for you), is that when running the application from the AppImage, the environment is not preserved, therefore Fedora doesn't configure the application to run correctly on top of XWayland.

I'm downloading Fedora 24, and we'll see.

jviotti, 2016-08-04:
Hi @GrzegorzBB ,

I've figured it out. You have to make sure you're loading the `xwayland.so` module in your configuration file. It works flawlessly for me with the following [weston.ini](http://manpages.ubuntu.com/manpages/wily/man5/weston.ini.5.html):

```
[core]
modules=xwayland.so
```

Since this is more about configuring Wayland correctly, I'll add a documentation entry in [USER-DOCUMENTATION.md](https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md) for it.

Please let me know if the problem still persists after the above suggestion.

closed 2016-08-04 by jviotti

GrzegorzBB, 2016-08-06:
You're the man Juan! ;) Everything works wine, thank you once more.

jviotti, 2016-08-07:
Haha I'm really glad it works for you, enjoy, and don't hesitate in pinging me if you encounter any other issue!