balena-io/etcher #531: Error EPERM

brandonhon, 2016-06-24:
No matter what I do I get an error when trying to use Etcher. Here is the error message I am getting:

EPERM
{"cause":{errno":-4048,"code":"EPERM","syscall":"write"},"isOperational":true,"errno":-4048,"code":....:"write"}

I am using Win8.1 64bit.

Jbithell, 2016-06-25:
[image]
I'm getting the error as well. 
Running Windows 10 64 bit, with 1.0.0-beta9

Jbithell, 2016-06-25:
Formatting the drive using the [SD Formatter Tool](https://www.sdcard.org/downloads/formatter_4/) solved the problem - not sure why Etcher couldn't do it

jviotti, 2016-06-27:
Hi @brandonhon , @Jbithell ,

Thanks for reporting this issue. We've previously seen this for the following reasons:

1. The SDCard is locked (although v1.0.0-beta.9 contains code to prevent a locked SDCard from being selected at all).
2. The drive you're trying to flash to was formatted with a GPT (GUID Partition Table) (only happens on Windows, but has been fixed recently).

We rely on the operating system to determine if the SDCard is locked, but of course, there is a chance the operating system is not recognising this correctly. Alternatively, the SDCard might be software-protected. Do you mind checking this? If this is the case, then this is a great opportunity to strengthen our write-protection detection logic.

Regarding GPT, the problem was that Windows only allows direct physical access to drives that contain no file system. For MBR devices, we could write zero bytes to the first sector of the drive, which was enough, however GPT devices store the partition table in more than one sector, and also duplicate the partition table at the end of the drive. We couldn't make sure the GPT was erased manually since our application could only write to the first sector (due to policies). As a workaround, we relied on `diskpart` `clean` command to clean the filesystem, which you can reproduce yourself by running `diskpart.exe`, determine the disk id with `list disk`, select the appropriate one with `select disk X`, and call `clean`.

Do you remember if the drive presented any of the above characteristics before flashing to it?

Zeirison, 2016-06-27:
[image]
Windows 10 x64, 1.0.0-beta9.
Exactly the same bug in Beta 9, in the previous version everything was fine. Sorry for my English.

jviotti, 2016-06-27:
Hm, the fact it happens at the beginning of the progress bar implies that we're violating Windows direct physical drive policies for some reason :/ The only thing I can think of is that we should trigger a drive rescan after erasing the file system with `diskpart`. I'll try adding that and some delay, and see if it fixes it. I'll upload a custom build shortly.

jviotti, 2016-06-27:
Can you try this build out: https://www.dropbox.com/s/sacqzg6vqjpy7wn/Etcher-win32-x64-eperm-fix-attempt.exe?dl=0?

Zeirison, 2016-06-27:
[image]
And so for several minutes.

jviotti, 2016-06-27:
Hi @Zeirison ,

Yeah, I've just discovered a small issue what made the application crash when the percentage was equal to zero.

I believe I have a definitive issue in this build: https://www.dropbox.com/s/uvz21fkvumd4nv7/Etcher-win32-x64.exe?dl=0

Zeirison, 2016-06-27:
[image]
Thanks, the problem was completely solved =)

jviotti, 2016-06-27:
Awesome! Thanks a lot everyone for submitting this issue and helping out testing the fix, very appreciated! I've submitted a PR with the fix, and will be releasing a new version with it probably today or very early tomorrow!

closed 2016-06-27 by jviotti