balena-io/etcher #532: osx 10.9 select image

morriswinkler, 2016-06-25:
For some reason i can not access files in my user folder on osx 10.9 only outside, if i select for exampel my downlaod folder i get an empty view, /Applications do work.

Any suggestions

jviotti, 2016-06-27:
Hi @morriswinkler ,

I believe you're hitting the following issue: https://github.com/resin-io/etcher/issues/287.

Until `v1.0.0-beta.9`, Etcher was completely ran as `root`, causing the open dialog to display `root`'s account common directories, rather than the one from your normal user. This can usually be workaround by manually navigating to `/Users/<your user>`, where you'll find the directories that belong to you.

The good news is that this issue was very recently fixed by https://github.com/resin-io/etcher/pull/423, which causes only the writer process to be elevated, causing the open dialog to be opened as your normal user. This feature will land on the next Etcher release, but at least for now, I've built a custom developer preview Etcher built containing this fix for your convenience. Let me know if this works for you!

Link: https://www.dropbox.com/s/73msk7mmv8mbzwf/Etcher-writer-elevate-preview-x64.dmg?dl=0

jviotti, 2016-07-07:
I'm closing this issue since it was fixed in v1.0.0-beta-10.

closed 2016-07-07 by jviotti