balena-io/etcher #552: Command failed and child process exited errors ...

nkolban, 2016-07-01:
I am running Etcher 1.0.0.0-beta-10 on a Windows 10 x64 machine.  When I am burning Raspbian images onto micro SD cards, I spuriously get a variety of errors.  Here are two screen shots of the different type of errors received:

[image]

and

[image]

The interesting thing (to me) is that the errors are non-deterministic.  If I simply press the "Flash" button a second time, it may work or I may get one of the above issues.   I have to keep trying until it works.

jviotti, 2016-07-04:
Hi @nkolban ,

Thanks for reporting this. The errors that you're getting are indeed very interesting. I'm mostly interested in the `diskpart` one, since the other provides little to no information to debug it.

We use `diskpart` to clean and rescan your drives (since Windows policies don't allow direct physical access to the devices with a filesystem beyond the first sector), so I don't have a clear idea of why `diskpart` could be failing.

Some things I'd try:

- Try plugging a drive (without nothing important on it), open `diskpart.exe` from `cmd.exe`, or the "Run" menu. Find the drive id with `list disk`, select it with `select disk X`, then `clean` and `rescan`. Maybe you get more information that way?

- Try reproducing the `diskpart.exe` bug, and while the error dialog is open (the file will get deleted once you close it, I believe), see if the script whose path is shown there actually exists, and has read permissions.

- Try other drives and see if it works fine there. Maybe the drive you were trying to write to has some weird issues that's causing `diskpart.exe` or the writer script to fail for some reason.

jmjeffers, 2016-07-12:
I get cscript errors on start up consistently.  I can only comment that I have a W10-64 machine with 14 partitions 10 being Linux Debian types ext4 and this is always in the start up process.  

[image]

jviotti, 2016-07-12:
Hi @jmjeffers ,

Thanks a lot for reporting. This is undoubtedly a weird issue we can't manage to reproduce here. Is `cscript` accessible from your path? I can't think of any reason `cscript` might not be accessible. Have you guys tweaked your `%PATH%` environment variable? What's the current value?

P.S: I see your comment came duplicated in GitHub for some reason, so I'm deleting one of them!

jviotti, 2016-07-12:
Another thought: I see you have Git Bash, Git Shell and other similar terminal emulators installed. Could it be that the command using `cscript` (we run a small VBScript to detect the available drives) is trying to get spawned from another shell, which does't have `cscript` in the path?

A way to check this would be to verify if `cscript` is not available in any of your shells, and tweak the paths to make it so. If my theory is correct, Etcher should run fine afterwards.

jviotti, 2016-07-18:
@nkolban , The initial two errors reported have been fixed in https://github.com/resin-io-modules/etcher-image-write/pull/46, and will land in v1.0.0-beta.11.

jviotti, 2016-07-18:
@jmjeffers, Is there a chance that Windows Script Host is disabled on that machine? (see http://www.thewindowsclub.com/windows-script-host-access-is-disabled-on-this-machine)

jviotti, 2016-08-07:
I'm closing this issue since the first two problems were resolved in v1.0.0-beta.13, and the remaining one seems to have fixed itself on the latest version as reported in https://github.com/resin-io/etcher/issues/628. We'll re-open if it ever happens again.

closed 2016-08-07 by jviotti

beachmiles, 2017-04-05:
Still getting a similar error on beta 19 on Win 10 x64,
[image]

lurch, 2017-04-05:
@beachmiles I think the error you're seeing is actually #989 - please follow up there :-)