balena-io/etcher #553: ESXi ISO fails to boot

waspinator, 2016-07-02:
1.0.0-beta. 10
OSX 10.11(.0)

ESXi ISO image does not boot after flashed to USB
`VMware-VMvisor-Installer-6.0.0.update02-3620759.x86_64.iso`

The flashing processes seems to go okay, no errors reported in the dev tools or otherwise.

Image is avaialble for free, but with registration at vmware.com

Drive was formatted to FAT32

[LinuxLive USB](http://www.linuxliveusb.com/) creates a bootable and working drive.

I noticed that when plugged into a PC, the USB drive created with etcher is not recognized (asks to be formatted), while the one created with LinuxLive looks like a regular usb drive with files on it.

jviotti, 2016-07-04:
Hi @waspinator ,

Thanks for reporting. I'll give the image a spin myself and will report back how it goes!

jviotti, 2016-07-04:
This VMWare image is very weird. It doesn't contain a partition table at all, which is the reason you can't open it in Windows (its simply a raw partition).

I can confirm Etcher is flashing the image the correct way, however the image seems to need special treatment to be made bootable (it doesn't work with `dd` neither). LinuxLive USB simply puts the `*.iso` as it is in the drive, and writes a special configuration file to boot to it, while UNetBootin for example, writes a GRUB to the drive, creating an entry that points to the partition.

I'm not sure what do to about this. The image works in other Linux specific writing tools since well, they are intended to write Linux images only, and therefore can make some assumptions, and do some extra stuff around that.

We could add support for these stuff on Etcher, but it might conflict with non-Linux images, so I guess it would come as an opt-in feature.

waspinator, 2016-07-04:
I doubt VMware would be interested in changing their ISO build process, but I guess it can't hurt to try and contact them.

If writing linux specific ISOs is not in etcher's mandate, then perhaps it can produce an error when someone tries to use an incompatible image. Maybe even suggest LinuxLive or unetbootin as an alternative when displaying the error.

jviotti, 2016-07-05:
I'd love to keep this issue in the backlog in order to think more about
it. I'm all for it if we find a way to introduce it a safe way.

What we can do for now is detect if the image you select contains a
partition table, and if not, present a warning message along the
following lines:

> WARNING: The image you selected doesn't appear to be valid (doesn't
> have a valid partition table).
> Please make sure that you're image is not corrupted, or check with the
> provider in case this image require special treatment to be made
> bootable, or requires special tooling to be flashed.

What do you think?

taahirisaacs, 2016-07-05:
@jviotti SGTM, I think a subtle warning alert should do the trick here

jviotti, 2016-07-05:
@taahirisaacs Do you think we can re-use/extend the orange warning ribbon that was implemented here: https://github.com/resin-io/etcher/issues/45#issuecomment-204415471?

taahirisaacs, 2016-07-05:
@jviotti I was thinking along those exact lines

[image]

jviotti, 2016-07-20:
@taahirisaacs We should still allow some way for the user to ignore the warning and proceed.

lurch, 2016-09-29:
> We should still allow some way for the user to ignore the warning and proceed.

Yeah. I recall back in the days of the early LiveUSB Linux distros (DamnSmallLinux etc.) there were some distros that used "hard drive" formatting (i.e. with a traditional MBR at the start of the disk) and some that used "superfloppy" formatting (i.e. just the raw partition written to the disk, with no MBR). I think at the time, that some BIOSes could only boot one or the other.

jviotti, 2017-03-21:
I'm closing this issue in favour of https://github.com/resin-io/etcher/issues/1189, which is concerned with detecting partition tables on drives.

closed 2017-03-21 by jviotti

lurch, 2017-05-08:
A user on Gitter reports that VMware esxi 6.5 has the same "problem".

brettveenstra, 2018-04-12:
still an issue with VMware vSphere Hypervisor 6.5.0a ISO, even after #1189 & #1248 work - any other update planned to support these media types?

gbdavidx, 2018-04-28:
can't boot either, same error as above on dell r510 server

lurch, 2018-05-03:
I wonder if Etcher could do any special-detection of VMWare images, and then suggest an alternative flashing tool, just like the Windows-images special behaviour Etcher has?

Mac-On, 2018-05-03:
At all my different clients, VMware vSphere/ESXi is "the standard" in enterprise environments – so if you wanna use a commercial virtualisation solution, normally you will use VMware. Saying this, it's not only 'another vendor' but a solution, which is well known and rock solid. Too sad the developers of Etcher won't support it, so better use unetbootin, which works fine out of the box, as I've already mentioned above.

doertedev, 2019-05-24:
https://github.com/balena-io/etcher/issues/553#issuecomment-386222985 << same problem with the UBCD and BOTH unetbootin and etcher from macos, with the exception that neither produce a bootable medium.

lurch, 2019-05-24:
> same problem with the UBCD

See also #2704

xqgex, 2022-11-07:
I can confirm that on Ubuntu 22.04 Etcher 1.5.115 cannot burn the latest VMware ESXI/vSphere 7.0 and unetbootin 702 can.

lurch, 2022-11-07:
> Etcher 1.5.115 cannot burn the latest VMware ESXI/vSphere 7.0 and unetbootin 702 can.

See https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#why-is-my-drive-not-bootable

ocroquette, 2023-10-26:
I was not successful with unetbootin-linux64-702.bin, but Ventoy did the trick.