balena-io/etcher #554: Allow terminal other than xterm

act78723, 2016-07-02:
- **What Etcher version are you running?**
1.0.0-beta.10 Etcher-linux-x64.AppImage

- **What operating system and architecture are you running Etcher in?**
LinuxMint (Mate x64) 17.3

- **What is the behaviour you're experiencing? How can we reproduce it?**

Details: Failed to execute child process "xterm" (No such file or directory)

  - [OS X](https://raw.githubusercontent.com/resin-io-modules/drivelist/master/scripts/darwin.sh)
  - [GNU/Linux](https://raw.githubusercontent.com/resin-io-modules/drivelist/master/scripts/l
(https://raw.githubusercontent.com/resin-io-modules/drivelist/master/scripts/linux.sh)
  - [Windows](https://raw.githubusercontent.com/resin-io-modules/drivelist/master/scripts/win32.bat)

jviotti, 2016-07-04:
Hi @act78723 ,

Thanks for reporting this! At which point does this error happen (e.g: when you run the AppImage, right after you click "Flash", etc)?

We're definitely not hardcoding a terminal name anywhere, so my guesses are either that the NodeJS Child Process module is picking `xterm` by default, or this comes from the AppImage runtime code.

act78723, 2016-07-04:
I believe it's coming from the NodeJS Child Process.

jviotti, 2016-07-05:
I'll give this a go on Ubuntu 16.04 after making sure `xterm` is removed
and see if I can reproduce.

jviotti, 2016-07-07:
Hi @act78723 ,

I've been trying to reproduce this one for a while without luck. I did encounter the following discussions about a very similar issue that happens exactly on the Mate desktop:

- https://github.com/mate-desktop/mate-panel/issues/57
- https://bugs.launchpad.net/ubuntu-mate/+bug/1415048
- http://askubuntu.com/questions/419641/panel-apps-try-to-launch-in-xterm-even-though-it-isnt-installed

The weird thing is that the bug mentioned in those discussions happens when you create a custom application launcher with "Application in terminal" type, which I highly doubt AppImages are doing.

Are you creating a custom launcher for Etcher, or are you running it with the image that was installed by the dialog presented when you opened Etcher for the first time?

jviotti, 2016-07-07:
Also, do you mind pasting the contents of your application launcher? It should be in `$HOME/.local/share/applications` or `/usr/local/share/applications` apparently.

jviotti, 2016-07-07:
Its indeed lacking `Terminal=true` in my system (Ubuntu 16.04 x64 w/Unity):

```
[Desktop Entry]
Name=Etcher
Exec=/home/parallels/Projects/etcher/etcher-release/installers/Etcher-linux-x64.AppImage
Comment=Burn images to SD cards & USB drives, safe & easy
Icon=/home/parallels/.cache/thumbnails/normal/62aff500cfa829ae140de5e723814ce2.png
Type=Application
X-Desktop-File-Install-Version=0.22
X-AppImage-Comment=Generated by ./bin//elevate.wrapper
TryExec=/home/parallels/Projects/etcher/etcher-release/installers/Etcher-linux-x64.AppImage
```

jviotti, 2016-07-07:
I'm downloading your exact Linux Mint version, we'll see in a minute...

act78723, 2016-07-08:
>   * mate-desktop/mate-panel#57

That is exactly how I'm running it. Custom launcher/app in terminal.

jviotti, 2016-07-08:
Hi @act78723 ,

Thanks for confirming that. There are plenty of workarounds described in the websites I pasted above, although the real fix would be to not use `Terminal=true` in your launcher, since its not necessary.

I'm closing this as Linux Mint known issue.

closed 2016-07-08 by jviotti

act78723, 2016-07-08:
Thank you for checking this out