balena-io/etcher #557: Etcher from source error in Debian

DiegoAccorinti, 2016-07-06:
- **What Etcher version are you running?**
v1.0.0-beta.10 - 2016-06-27

- **What operating system and architecture are you running Etcher in?**
Linux huayra 3.16.0-4-amd64 #1 SMP Debian 3.16.7-ckt25-2 (2016-04-08) x86_64 GNU/Linux

- **What is the behaviour you're experiencing? How can we reproduce it?**
I clone the repo, do "npm install" : 

```
npm WARN package.json JSV@4.0.2 'repositories' (plural) Not supported.
npm WARN package.json Please pick one as the 'repository' field
npm WARN package.json eyes@0.1.8 No repository field.
npm WARN package.json get-package-info@0.0.2 No repository field.
npm WARN package.json indexof@0.0.1 No repository field.
npm ERR! notsup Unsupported
npm ERR! notsup Not compatible with your operating system or architecture: macmount@1.0.0
npm ERR! notsup Valid OS:    darwin
npm ERR! notsup Valid Arch:  any
npm ERR! notsup Actual OS:   linux
npm ERR! notsup Actual Arch: x64

npm ERR! System Linux 3.16.0-4-amd64
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install"
npm ERR! cwd /home/alumno/programas/etcher
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! code EBADPLATFORM
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/alumno/programas/etcher/npm-debug.log
npm ERR! not ok code 0
```

and when execute "electron ."  The app  starts empty.

[image]

**DevTools LOG:**

```
Uncaught Error: Module did not self-register.
Uncaught Error: [$injector:modulerr] Failed to instantiate module Etcher due to:
Error: [$injector:nomod] Module 'Etcher' is not available! You either misspelled the module name or forgot to load it. If registering a module ensure that you specify the dependencies as the second argument.
http://errors.angularjs.org/1.5.7/$injector/nomod?p0=Etcher
    at /home/alumno/programas/etcher/node_modules/angular/angular.js:68:12
    at /home/alumno/programas/etcher/node_modules/angular/angular.js:2075:17
    at ensure (/home/alumno/programas/etcher/node_modules/angular/angular.js:1999:38)
    at module (/home/alumno/programas/etcher/node_modules/angular/angular.js:2073:14)
    at /home/alumno/programas/etcher/node_modules/angular/angular.js:4608:22
    at forEach (/home/alumno/programas/etcher/node_modules/angular/angular.js:321:20)
    at loadModules (/home/alumno/programas/etcher/node_modules/angular/angular.js:4592:5)
    at createInjector (/home/alumno/programas/etcher/node_modules/angular/angular.js:4514:19)
    at doBootstrap (/home/alumno/programas/etcher/node_modules/angular/angular.js:1751:20)
    at bootstrap (/home/alumno/programas/etcher/node_modules/angular/angular.js:1772:12)
    at angularInit (/home/alumno/programas/etcher/node_modules/angular/angular.js:1657:5)
    at /home/alumno/programas/etcher/node_modules/angular/angular.js:31468:5
    at HTMLDocument.trigger (/home/alumno/programas/etcher/node_modules/angular/angular.js:3198:7)
    at defaultHandlerWrapper (/home/alumno/programas/etcher/node_modules/angular/angular.js:3488:11)
    at HTMLDocument.eventHandler (/home/alumno/programas/etcher/node_modules/angular/angular.js:3476:9)
http://errors.angularjs.org/1.5.7/$injector/modulerr?p0=Etcher&p1=Error%3A%…umno%2Fprogramas%2Fetcher%2Fnode_modules%2Fangular%2Fangular.js%3A3476%3A9)

```
Tnks for help.

jviotti, 2016-07-06:
Hi @DiegoAccorinti ,

Thanks for reaching out! Sadly, installing Etcher's dependencies is not as simple as doing an `npm install`. The whole process of configuring NPM to make use of Electron's v8 runtime when compiling dependencies is outlined in our CONTRIBUTING guide: https://github.com/resin-io/etcher/blob/master/docs/CONTRIBUTING.md#running-locally.

In summary:

- You'll need to have the *exact* NodeJS version being used by the current Etcher's Electron version to avoid binary incompatibility (which at the time is v6.1.0).

- You'll need to configure `npm` with the following environment variables:

```
export npm_config_disturl=https://atom.io/download/atom-shell
export npm_config_target=<ELECTRON_VERSION>
export npm_config_runtime=electron
```

At the time of this writing, the electron version we use if 1.1.1.

- You'll need to run `npm install` with the `--build-from-source` option, to avoid pre-compiled packaged from being downloaded.

- You'll need to install bower dependencies as well.

Let me know if you still get stuck in any of the above steps!

DiegoAccorinti, 2016-07-06:
Hi @jviotti , tnks for respond.

> You'll need to have the exact NodeJS version being used by the current Etcher's Electron version 
> to avoid binary incompatibility (which at the time is v6.1.0).

i downloaded and installed 6.0  , don't found 6.1  at [https://nodejs.org/download/nightly/](https://nodejs.org/download/nightly/)

> You'll need to configure npm with the following environment variables

OK

> You'll need to run npm install with the --build-from-source option, to avoid pre-compiled packaged from being downloaded.

OK !

> You'll need to install bower dependencies as well.

OK !

only rest:

alumno@huayra:~/programas/etcher$ **bower install**
alumno@huayra:~/programas/etcher$ **electron .**

and it's READY.

[image]

tnks very much!
Diego.

closed 2016-07-06 by DiegoAccorinti

jviotti, 2016-07-06:
Hi @DiegoAccorinti ,

I'm glad to hear you got it working! Enjoy, and don't hesitate to speak to us in out Gitter chat channel (https://gitter.im/resin-io/etcher) if there's anything else we can do for you.