balena-io/etcher #567: Invalid image

poohomar, 2016-07-11:
[image]

- **What Etcher version are you running?** - 1.0.0-beta.10

*(You can check this at runtime in the bottom right corner of the application. If you're running Etcher locally, please let us know in which commit or branch you're in.)*

- **What operating system and architecture are you running Etcher in?** - Windows 7 x64 Enterprise

*(For example: Windows 10 x64, OS X 10.11.4 x64, Ubuntu 14.04 x86.)*

- **What is the behaviour you're experiencing? How can we reproduce it?** Yes, it keeps on telling the image is invalid, but it's in .iso format.

- **Do you have any input on how this can be fixed/improved?** N/A

- **Was this happening on a previous version?** N/A

- **Do you see any meaningful error information on DevTools?** N/A

*(You can open DevTools by pressing `Ctrl+Alt+I`, or `Cmd+Alt+I` if you're running OS X.)*

***

**If you're having an error when flashing an image to a drive: **

*(Feel free to delete this whole section if it doesn't apply.)*

- **Can you reproduce the issue with a different image or drive?**

- **Can you share a link to the exact image you're trying to flash, so we can test it ourselves?**

- **At which point of the flashing does the error happens? (right at the beginning, randomly, etc)**

- **What was the format of the drive _before_ attempting to flash to it?**

***

**If Etcher is failing to detect your drive: **

*(Feel free to delete this whole section if it doesn't apply.)*

- **Can you post the output of the script that applies to your operating system from the following list?:**

  - [OS X](https://raw.githubusercontent.com/resin-io-modules/drivelist/master/scripts/darwin.sh)
  - [GNU/Linux](https://raw.githubusercontent.com/resin-io-modules/drivelist/master/scripts/linux.sh)
  - [Windows](https://raw.githubusercontent.com/resin-io-modules/drivelist/master/scripts/win32.bat)

jviotti, 2016-07-11:
Hi @poohomar ,

Thanks for reporting. I believe the issue is that the file extension is in uppercase, causing the check to fail. I'll push a fix in a minute. For now, renaming to lowercase should suffice as a workaround.

closed 2016-07-11 by jviotti

poohomar, 2016-07-15:
Thanks for the fast response, all is well now and appreciate the knowledge!

fremindmm, 2018-05-11:
Dowload this image https://downloads.lede-project.org/releases/17.01.4/targets/brcm2708/bcm2710/lede-imagebuilder-17.01.4-brcm2708-bcm2710.Linux-x86_64.tar.xz
Mac OS   1.4.4 etcher 
invalid image error:
[image]

jhermsmeier, 2018-05-14:
@fremindmm tarballs aren't supported yet, you'll have to extract the image beforehand

jhermsmeier, 2018-05-14:
I've added [an issue](https://github.com/resin-io/etcher/issues/2338) to track tarball support

lurch, 2018-05-14:
@jhermsmeier Even if Etcher _did_ support tarballs, I don't think it'd be able to flash the 'image' that @fremindmm has downloaded...
[image]

@fremindmm I did a bit of digging, and I suspect the file you _actually_ want is https://downloads.openwrt.org/releases/17.01.4/targets/brcm2708/bcm2710/lede-17.01.4-brcm2708-bcm2710-rpi-3-ext4-sdcard.img.gz which Etcher should support already.