balena-io/etcher #573: Validation hangs when the disk is too small

tprk77, 2016-07-17:
- **Etcher version:** 1.0.0-beta.10
- **Operating system and architecture:** Windows 7, x86_64
- **Do you see any meaningful error information on DevTools?** No

I'm trying to burn an image from the `ev3dev` project, [`ev3-ev3dev-jessie-2015-12-30.img.xz`](https://github.com/ev3dev/ev3dev/releases/download/ev3dev-jessie-2015-12-30/ev3-ev3dev-jessie-2015-12-30.img.xz). Not sure if that matters.

With a 4 GB card, validation hangs at `Progress (check): 99% at 17.99 MB/s (eta 1s)`. The button was stuck on `Finishing...` for at least 30 minutes. There was also a long pause at the end of writing. I tried two different cards. Please let me know if I can provide any other useful information.

With a 16 GB card, there was no problem.

I would expect Etcher to display an error message saying the disk was too small. Hopefully before it starts burning, to avoid wasting time.

tprk77, 2016-07-17:
The original title was a bit of a guess, based on the three cards I was using. Let me know if I should try anything else. Thanks!

jviotti, 2016-07-18:
Hi @tprk77 ,

Thanks a lot for reaching out. I believe this is somewhat related to https://github.com/resin-io/etcher/issues/571, where another user reports than the same image gets stuck at 80%, presumably because of a small drive as well.

I'd initially guess that the image is not being compressed correctly (and therefore the decompression module hangs), but the image looks ok, and you both report that is finishes successfully with a bigger drive.

Can you test uncompressing the image on Windows first (using something like 7zip), and attempting to burn the uncompressed image to that 4 GB drive?

jviotti, 2016-07-25:
I discard the assumption of the SDCard being too small given this error happens during the check phase, where we only read back data. When the check process reaches "Finishing...", its because its waiting for the drive to auto-unmount before reporting the final "success" screen, so I bet there is something causing `umount` wait forever.

tprk77, 2016-07-27:
Sorry I never tried extracting with 7zip, should I still do that? Do you want me to try anything else?

tprk77, 2016-07-27:
So I tried again with `1.0.0-beta.11` and a 2 GB card, and surprisingly it worked! I thought that was weird, so I went back to  `1.0.0-beta.10` with the 2 GB card, and even more surprisingly, that worked too. Then I tried the 4 GB card, and... that didn't work. I closed the app by pressing the X. I tried the 2 GB card again, and it also didn't work, it didn't even start.

Not sure how to explain any of that. It seems that somehow Etcher or Windows is getting into a bad state? Let me try a reboot.

tprk77, 2016-07-27:
Well, `1.0.0-beta.11` seems to work every time, so we can probably close this.

jviotti, 2016-07-27:
Hi @tprk77 ,

Thanks a lot for the testing. This is indeed really strange. I sadly can't reproduce this on my Windows 10 PC to provide any input. I do have an old Windows 7 one around, so I'll give it a go there to see what happens.

The only thing that comes to my mind is that some software might be trying to do I/O with the drives and randomly interfering Etcher. Maybe an anti-virus, or some sort of drive scanner daemon? Some applications also try to load files from removable drives (think photography or DAW software), and keep a process opened causing other applications to get EBUSY errors when trying to unmount, or any other operation on it.

jviotti, 2016-07-27:
Hi @tprk77 ,

Glad to hear that! I still suspect this might still happen (I can't see how v1.0.0-beta.11 could have solved this, but maybe it did), so please re-open if so!

checktravis, 2016-07-27:
i just installed 1.0.0-beta 12 and it's hanging after validation with the "finishing" message on the write button. it's been about 30 mins, i'm ending the task and seeing if the card works (since it looks like validation finished)

jviotti, 2016-07-27:
Hi @checktravis ,

Thanks a lot for the report. I suspected this would happen again, so the fact is started working for @tprk77 might be just a coincidence.

The card should have been written successfully, so I believe I should boot just fine.

I insist that there is something wrong when attempting to unmount the drive on Windows. Can you try disabling the "auto-unmount on success" feature from the settings page (the cog in the top right corner) and see if it still happens? If you confirm it works when you disable it, but then happens again if you enable it, we would effectively narrow the problem to the un-mounting routine. 

Also, can you try stopping any applications or services that could be scanning your drives and give it another go?

We use `RemoveDrive.exe` from http://www.uwe-sieber.de/drivetools_e.html to handle removal, so there is also the possibility that this tool is not working appropriately in some cases. Another thing to try would be to download the tool and see if it works fine for you.

jviotti, 2016-07-27:
There are actually various `RemoveDrive.exe` command line options that could help fixing the issue:

```
[-L]       loop until success
[-e]       try to dismount and eject if the removal fails
```

Maybe we can get some more diagnosis from:

```
[-dbg]     show debug information
```

I'll make some custom builds with the above arguments so we can test if it makes any difference.

jviotti, 2016-07-27:
Great news! I can reproduce this on my old Windows 7 PC, so seems to be a Windows 7 specific issue. I'll be investigating more today and will report back how it went.

jviotti, 2016-07-28:
After a lot of investigation, I believe I know what's happening, and hopefully have a fix for it. 

Right after the image is flashed (before the validation phased), the drive is re-mounted and assigned a drive letter. The unmount routine takes the drive letter and attempts to unmount the drive. The problem is that Windows doesn't automatically acknowledge that a new drive letter was assigned to the drive, and when we pass it to `RemoveDrive.exe`, this program exits complaining that the drive letter doesn't exist.

The fix is to force Windows to rescan the available drives right before attempting to unmount, which seems to fix the issue.

I'll send a PR soon, and will upload a custom build so that you can confirm the fix. In the mid-time, disabling the auto-unmounting on success indeed evades the issue.

jviotti, 2016-07-28:
We're getting close. I'm now seeing a new issue where the unmounting routine would fail when the validation check is enabled due to the drive remaining busy.

jviotti, 2016-07-28:
I've fixed the drive busy error by making sure the device stream file descriptor is closed after the validation phase before returning control to the caller.

Now that all the "issues" are fixed, I'm experiencing some problems with the tailing module, which fails to emit the "done" event back to the application in some cases for some reasons. This might be the cause of the application getting stuck at the end of a flash process without any apparent error.

Overall it looks like the Windows issues reported lately to Etcher were caused by many unrelated problems showing up at once. Let's hope this is the last issue I find :)

jviotti, 2016-08-03:
After the following PRs, looks like all Windows 7 issues are gone:

- https://github.com/resin-io/etcher/pull/616
- https://github.com/resin-io-modules/etcher-image-write/pull/46
- https://github.com/resin-io-modules/etcher-image-write/pull/48

I can confirm on Windows 7 that checking out an Etcher reference which contains the above merged PRs makes the issues disappear, and going back in history makes the issues visible again. For that reason, I'll be closing this issue.

Please re-open if you still hit this issue once the a Etcher version is released.

Thanks a lot!

closed 2016-08-03 by jviotti