balena-io/etcher #574: Disk auto-mounts partway through validation on Xubuntu 16.04

tprk77, 2016-07-17:
- **Etcher version:** 1.0.0-beta.10
- **Operating system and architecture:** Xubuntu 16.04, 4.4.0-31-generic, x86-64
- **Do you see any meaningful error information on DevTools?** No

Trying to work around #573, I switched over to Linux and a different card writer. I've noticed that on this system, after the write finishes, the partitions auto-mount and pop-up the file manager. This happens partway through the validation phase, which I believe is causing it to fail. (So far, validation always fails on this machine.)

I think the auto-mount is changing the disk, because I can do the same thing with `dd` and see the checksum change. For example:

```
t@comp:~/t$ sudo dd bs=4M if=ev3-ev3dev-jessie-2015-12-30.img of=/dev/sdd
450+0 records in
450+0 records out
1887436800 bytes (1.9 GB, 1.8 GiB) copied, 162.548 s, 11.6 MB/s
450+0 records in
450+0 records out
1887436800 bytes (1.9 GB, 1.8 GiB) copied, 162.548 s, 11.6 MB/s
t@comp:~/t$ sync
t@comp:~/t$ md5sum ev3-ev3dev-jessie-2015-12-30.img
4d25e323fbde509c9d5869cf77a513cc  ev3-ev3dev-jessie-2015-12-30.img
t@comp:~/t$ sudo dd bs=4M if=/dev/sdd count=450 | md5sum
450+0 records in
450+0 records out
1887436800 bytes (1.9 GB, 1.8 GiB) copied, 85.9326 s, 22.0 MB/s
4d25e323fbde509c9d5869cf77a513cc  -
t@comp:~/t$ # Reinsert locked card, auto-mount
t@comp:~/t$ sudo dd bs=4M if=/dev/sdd count=450 | md5sum
450+0 records in
450+0 records out
1887436800 bytes (1.9 GB, 1.8 GiB) copied, 85.7719 s, 22.0 MB/s
4d25e323fbde509c9d5869cf77a513cc  -
t@comp:~/t$ # Reinsert unlocked card, auto-mount
t@comp:~/t$ sudo dd bs=4M if=/dev/sdd count=450 | md5sum
450+0 records in
450+0 records out
1887436800 bytes (1.9 GB, 1.8 GiB) copied, 85.7621 s, 22.0 MB/s
3a0b5857c817bb97028b6aceea4a22fb  -
t@comp:~/t$ 
```

tprk77, 2016-07-17:
I figured out how to turn off auto-mount, and now Etcher is working as expected. It was a setting in the Thunar file manager. It would be nice if Etcher could temporarily suppress auto-mount, or at least warn the user to disable it manually.

jviotti, 2016-07-18:
Hi @tprk77 ,

Thanks a lot for reporting this as well. We take measures to prevent auto-mount in Windows and OS X, however we never encountered this problem on a GNU/Linux system.

I'll investigate if there is an easy way to prevent auto-mounting in this operating system and report back.

jviotti, 2016-08-14:
I have a theory about a small fix that would prevent this issue. At https://github.com/resin-io-modules/etcher-image-write, the user passes the path to the drive as an argument. We then open a readable stream on it to flash the image. Once the image is flashed, the file descriptor is closed, and we then re-open it again to perform the validation. The fact that the file descriptor gets closed is what seems to trigger the auto-mount. As a fix, we could manually open a file descriptor on the drive using `fs.open()`, and re-use that file descriptor for both the flashing and validation process using `fs.createReadStream(null, { fd })`.

jviotti, 2016-08-14:
Also, this should allow us to get rid of `denymount` and actually split the write and validation routine into two separate functions that take a file descriptor as an argument.

jviotti, 2016-08-23:
Looks like my fd idea worked fine :) I've sent a PR for it, and it will come on the next Etcher release. Thanks a lot for the report!

closed 2016-08-23 by jviotti

tprk77, 2016-08-23:
Thanks for your work on this!