balena-io/etcher #589: "Programming error" when trying to flash image

thbar, 2016-07-23:
- **Etcher version:** 1.0.0-beta.11
- **Operating system and architecture:** OS X El Capitan 10.11.4 (15E65)
- **Do you see any meaningful error information on DevTools?** I see "Flash error" (see below for full log).

While burning the latest [ev3 image](http://www.ev3dev.org/), I installed Etcher and tried to use it to burn the content to a MicroSD card, and I repeatedly get the following error:

[image]

And after closing the modal, I get:

[image]

When opening DevTools, I see this (saved to file here)

```
Sat Jul 23 2016 22:32:30 GMT+0200 (CEST) Application start
Sat Jul 23 2016 22:32:30 GMT+0200 (CEST) Checking for updates
Sat Jul 23 2016 22:32:34 GMT+0200 (CEST) Select image ({"path":"/Users/thbar/Downloads/ev3-ev3dev-jessie-2015-12-30.img.xz","size":118914544})
Sat Jul 23 2016 22:34:05 GMT+0200 (CEST) Flash ({"image":"/Users/thbar/Downloads/ev3-ev3dev-jessie-2015-12-30.img.xz","device":"/dev/disk2"})
Sat Jul 23 2016 22:34:22 GMT+0200 (CEST) Flash error
Objectmessage: "Programming error"__proto__: Object__defineGetter__: __defineGetter__()__defineSetter__: __defineSetter__()__lookupGetter__: __lookupGetter__()__lookupSetter__: __lookupSetter__()constructor: Object()hasOwnProperty: hasOwnProperty()isPrototypeOf: isPrototypeOf()propertyIsEnumerable: propertyIsEnumerable()toLocaleString: toLocaleString()toString: toString()valueOf: valueOf()get __proto__: __proto__()set __proto__: __proto__() undefined
 ```

jviotti, 2016-07-24:
Hi @thbar ,

Thanks for reporting this really weird issue. The "Programming error" error comes from [lzma-native](https://github.com/addaleax/lzma-native), I believe. I'll try to reproduce and report back my results.

In the mid-time, I'm pretty sure that uncompressing the file with `xz -d <file>.xz` and burning the `.img` directly should work for you.

jviotti, 2016-07-24:
Ok, I know what's going on. Turns out I didn't configure `npm` correctly when releasing v1.0.0-beta.11 for OS X (to make native libraries build agains the V8 version Electron is using). I apologise for that! I'll re-publish v1.0.0-beta.11 for OS X in a minute, and will ping you once that's ready.

Again, sorry for the inconveniences! Luckily, [this commit](https://github.com/resin-io/etcher/commit/ef96ae86a0422e41f935e14d0939a2b1e1518598) will make sure I never make the same mistake for OS X.

jviotti, 2016-07-24:
I've re-built and re-uploaded v1.0.0-beta.11 for OS X and I confirm it works. Please re-open if you still encounter the issue. Thanks a lot!

closed 2016-07-24 by jviotti

thbar, 2016-07-25:
@jviotti I tried again today and can confirm the error does not appear anymore. Thanks for your update and quick response!

jviotti, 2016-07-25:
I'm really happy it worked for you too. Thanks a lot for the heads up and have a nice day!