balena-io/etcher #609: Windows 10 x64 the program gives an error while writing

Zeirison, 2016-07-28:
- v1.0.0-beta.12
- Windows 10 x64
- Yes, here are the logs:
```
tracker.js:10 Error: Child process exitted with error code: 1
    at ChildProcess.child.on (C:\Program Files (x86)\Etcher\resources\app.asar\lib\src\child-writer\index.js:114:38)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at maybeClose (internal/child_process.js:850:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:215:5) undefineda.(anonymous function) @ tracker.js:10(anonymous function) @ C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:13708(anonymous function) @ C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:10347(anonymous function) @ C:\Program Files (x86)\Etcher\resources\app.asar\lib\gui\modules\analytics.js:85processQueue @ C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:16178(anonymous function) @ C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:16186$eval @ C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:17444$digest @ C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:17257(anonymous function) @ C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:17483completeOutstandingRequest @ C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:5955(anonymous function) @ C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:6234d.util.hasFunction.f.(anonymous function) @ tracker.js:3
```
When attempts to write, there are two following errors (They appear at random, no patterns):
[image]
[image]

jviotti, 2016-07-28:
Hi @Zeirison ,

Thanks a lot for reporting. I've sent a fix for the `Invalid results errorCode: 1` issue. We're still investigating the previous one, but couldn't find the root cause yet.

For some more information:

- https://github.com/resin-io/etcher/issues/552
- https://github.com/resin-io/etcher/issues/571

jviotti, 2016-08-03:
After the following PRs, looks like all Windows 7 issues are gone:

- https://github.com/resin-io/etcher/pull/616
- https://github.com/resin-io-modules/etcher-image-write/pull/46
- https://github.com/resin-io-modules/etcher-image-write/pull/48

I can confirm on Windows 7 that checking out an Etcher reference which contains the above merged PRs makes the issues disappear, and going back in history makes the issues visible again. For that reason, I'll be closing this issue.

Please re-open if you still hit this issue once the a Etcher version is released.

Thanks a lot!

closed 2016-08-03 by jviotti