balena-io/etcher #613: Win7 x64 Child Press exited with error code: 1

GeorgeIoak, 2016-08-02:
- **Etcher version:** : 1.0.0-beta12
- **Operating system and architecture:** Win7 x64
- **Do you see any meaningful error information on DevTools?**

Mon Aug 01 2016 17:53:31 GMT-0700 (Pacific Daylight Time) Progress (check): 99% at 17.01 MB/s (eta 2s)
tracker.js:10 Mon Aug 01 2016 17:53:31 GMT-0700 (Pacific Daylight Time) Progress (check): 99% at 17.01 MB/s (eta 1s)
tracker.js:10 Mon Aug 01 2016 17:53:32 GMT-0700 (Pacific Daylight Time) Progress (check): 99% at 17.01 MB/s (eta 1s)
tracker.js:10 Mon Aug 01 2016 17:53:40 GMT-0700 (Pacific Daylight Time) Flash error
tracker.js:10 Error: Child process exitted with error code: 1
    at ChildProcess.child.on (C:\Program Files (x86)\Etcher\resources\app.asar\lib\src\child-writer\index.js:114:38)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at maybeClose (internal/child_process.js:850:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:215:5) undefineda.(anonymous function) @ tracker.js:10
694 console messages are not shown.

This happened twice using the same 8GB SD card writing 2016-05-27-raspbian-jessie image

jviotti, 2016-08-02:
Hi @GeorgeIoak ,

Thanks a lot for reporting. I've been chasing this issue really hard on https://github.com/resin-io/etcher/issues/571. The issue is caused by a combination of related problems related to the disk unmount routine.

I fixed most of it by now, and will hopefully have it all covered for the next released. As a workaround, disabling the "unmount on success" feature from the settings screen (the cog at the top right corner) should prevent the error.

Let me know how it goes!

GeorgeIoak, 2016-08-02:
Hi @jviotti ,

Thank you for the prompt rely. I saw that thread but it was reported at the bottom to be corrected so I thought I would open a new issue. I'd like to report that I can confirm that changing the setting to disable "unmount on success" appears to have worked because no error was thrown.

Thanks,
George

P.S. I'll report back if the image is not bootable

jviotti, 2016-08-02:
Hi @GeorgeIoak ,

I'm glad to hear that worked for you. Don't hesitate in contacting us if you experience any other weird behaviour. Luckily, unmounting on Windows 7 will be working like a charm on the next version :)

jviotti, 2016-08-03:
After the following PRs, looks like all Windows 7 issues are gone:

- https://github.com/resin-io/etcher/pull/616
- https://github.com/resin-io-modules/etcher-image-write/pull/46
- https://github.com/resin-io-modules/etcher-image-write/pull/48

I can confirm on Windows 7 that checking out an Etcher reference which contains the above merged PRs makes the issues disappear, and going back in history makes the issues visible again. For that reason, I'll be closing this issue.

Please re-open if you still hit this issue once the a Etcher version is released.

Thanks a lot!

closed 2016-08-03 by jviotti