balena-io/etcher #626: Failed flash on windows 10

curcuz, 2016-08-05:
- **Etcher version:** 1.0.0-beta.12
- **OS:** Windows 10 Home Build 14393.10
- **Etcher DevTools:**

```
Fri Aug 05 2016 19:17:56 GMT+0200 (W. Europe Daylight Time) Progress (check): 99% at 11.63 MB/s (eta 1s)
tracker.js:10 Error: Invalid results errorCode: 1
    at storeReducer (C:\Program Files (x86)\Etcher\resources\app.asar\lib\gui\models\store.js:210:15)
    at Object.dispatch (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\redux\lib\createStore.js:179:22)
    at unsetFlashingFlag (C:\Program Files (x86)\Etcher\resources\app.asar\lib\gui\modules\image-writer.js:124:11)
    at performWrite.then.catch (C:\Program Files (x86)\Etcher\resources\app.asar\lib\gui\modules\image-writer.js:253:12)
    at processQueue (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:16170:28)
    at C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:16186:27
    at Scope.$eval (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:17444:28)
    at Scope.$digest (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:17257:31)
    at C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:17483:26
    at completeOutstandingRequest (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:5955:10)
    at C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:6234:7
    at d.util.hasFunction.f.(anonymous function) (file:///C:/Program%20Files%20(x86)/Etcher/resources/app.asar/node_modules/trackjs/tracker.js:3:492) undefineda.(anonymous function) @ tracker.js:10
tracker.js:10 Fri Aug 05 2016 19:18:07 GMT+0200 (W. Europe Daylight Time) Flash error
tracker.js:10 Error: Invalid results errorCode: 1
    at storeReducer (C:\Program Files (x86)\Etcher\resources\app.asar\lib\gui\models\store.js:210:15)
    at Object.dispatch (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\redux\lib\createStore.js:179:22)
    at unsetFlashingFlag (C:\Program Files (x86)\Etcher\resources\app.asar\lib\gui\modules\image-writer.js:124:11)
    at performWrite.then.catch (C:\Program Files (x86)\Etcher\resources\app.asar\lib\gui\modules\image-writer.js:253:12)
    at processQueue (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:16170:28)
    at C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:16186:27
    at Scope.$eval (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:17444:28)
    at Scope.$digest (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:17257:31)
    at C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:17483:26
    at completeOutstandingRequest (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:5955:10)
    at C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\angular\angular.js:6234:7
    at d.util.hasFunction.f.(anonymous function) (file:///C:/Program%20Files%20(x86)/Etcher/resources/app.asar/node_modules/trackjs/tracker.js:3:492) undefineda.(anonymous function) @ tracker.js:10
```
**Error Alert:**
[image]

**Notes**
* I was using an external USB adapter, which was plugged on the laptop via a USB type-c HUB
* I was trying to flash a standard resin.io base image

jviotti, 2016-08-05:
Hi @curcuz ,

Can you check if this still happens on the newly released v1.0.0-beta.13? That version should contain a fix to this issue.

curcuz, 2016-08-08:
@jviotti v1.0.0-beta.13 fixes this issue

closed 2016-08-08 by curcuz

jviotti, 2016-08-08:
Glad to hear, thanks a lot!