balena-io/etcher #627: Failed Write in Windows 10

wsloand, 2016-08-07:
- **Etcher version:** 1.0.0-beta.13
- **Operating system and architecture:** Windows 10 64-bit
- **Do you see any meaningful error information on DevTools?** I restarted etcher before trying to submit the bug.  On restart, it worked.

[image]

jviotti, 2016-08-08:
Hi @wsloand ,

Thanks a lot for reporting this issue. I honestly have no idea what could be causing such access denied errors, which clearly come from outside Etcher. Some thoughts:

- Could it be that some software/service running in your PC is actively trying to scan a drive's content? Examples of this could be anti-virus, DAW, professional photography software, etc. My theory is that there is a program trying to scan the files in your drive, causing the access denied error for any other process trying to write to it, however cleaning the drive with diskpart (which causes the drive to lose any file-system) fixes the issue since the programs don't see a mountable partition, and therefore don't try to access the drive. Maybe you can try re-formating the drive as FAT32 for example, and see if you can reproduce the issue again? If so, try temporarily disabling/closing any suspicious app and see if it suddenly starts working.

- Since diskpart says `See the System Event Log for more information.`, it might be worth diving there to see if we find anything worth investigating. Navigating the event log is usually not very friendly, and given the huge amount of stuff being logged there, it might be hard to identify anything worth looking (if you remember more or less the day and hour you ran Etcher, that could help narrowing your search). There is a tutorial to navigate the event log here, just in case: https://technet.microsoft.com/en-us/library/aa996634(v=exchg.65).aspx

wsloand, 2016-08-20:
I just had another similar error, and now I'm able to fully debug it and provide more information.

The flash disk had a pre-existing linux partition on it.  Without doing anything else, I tried to flash it (with the current Raspbian image, 2016-05-27-raspbian-jessie-lite.zip from https://www.raspberrypi.org/downloads/raspbian/).  I got the same error as above.

I looked in the event log, and I saw the following error after the Etcher run.  When I ran "diskpart", "select disk 1", "clean", I got the same error again.  When I ran "clean" a second time in diskpart, I got no error, and it said that "DiskPart succeeded in cleaning the disk."

Syslog:

> Log Name:      System
> Source:        VDS Basic Provider
> Date:          8/20/2016 5:00:23 PM
> Event ID:      5
> Task Category: None
> Level:         Error
> Keywords:      Classic
> User:          N/A
> Computer:      adams
> Description:
> Cannot zero sectors on disk \\?\PhysicalDrive1. Error code: 5@0101000F
> Event Xml:
> <Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
>   <System>
>     <Provider Name="VDS Basic Provider" />
>     <EventID Qualifiers="49664">5</EventID>
>     <Level>2</Level>
>     <Task>0</Task>
>     <Keywords>0x80000000000000</Keywords>
>     <TimeCreated SystemTime="2016-08-20T21:00:23.355617000Z" />
>     <EventRecordID>6342</EventRecordID>
>     <Channel>System</Channel>
>     <Computer>redacted</Computer>
>     <Security />
>   </System>
>   <EventData>
>     <Data>\\?\PhysicalDrive1</Data>
>     <Data>5@0101000F</Data>
>   </EventData>
> </Event>
>

wsloand, 2016-08-20:
After that and re-running the flash, I got a warning from Etcher that says "Your removable drive may be corrupted.  Try inserting a different one and press retry".  This happened after writing and appeared to happen during the verify stage (but I wasn't watching).  The logging output (Ctrl-Alt-I) showed up to "Progress (write): 98% at 30.26 MB/s" then went up to 98% check ending with these two messages:

> Sat Aug 20 2016 17:13:01 GMT-0400 (Eastern Daylight Time) Progress (check): 98% at 26.4 MB/s (eta 1s)
> tracker.js:10 Sat Aug 20 2016 17:13:07 GMT-0400 (Eastern Daylight Time) Validation error

The validation error appeared to coincide with the same system log error as above.

I retried with the same device, and it started in the logging output with "Sat Aug 20 2016 17:18:04 GMT-0400 (Eastern Daylight Time) Restart after failure".  That retry went without a problem.

jviotti, 2016-08-21:
Hi @wsloand ,

Thanks a lot for the extra information. We do try to re-run `clean` if it fails [up to 5 times](https://github.com/resin-io-modules/etcher-image-write/blob/master/lib/win32.js#L67), so I wonder why its still failing.

The validation error might be related, and is caused when the checksum of what was written to the drive doesn't equal the original image checksum.

Can you actively make the issue re-appear if you format your drive using the same Linux partitions as before?

jviotti, 2016-08-22:
Hi @wsloand ,

Just had a crazy thought: we retry the whole `diskpart` routine if it fails (e.g: start `diskpart`, select drive, etc). Maybe that causes `clean` to keep failing, but just retrying `clean` works.

Can you try retrying the whole thing (`diskpart.exe`, `select disk 1`, `clean`) and see if that fails?

Bottersnike, 2016-08-23:
Having just had the same error, I retried the img burn and it worked perfectly without even restarting Etcher (let alone using diskpart).

jviotti, 2016-08-23:
@Bottersnike So you get this error in the middle of a writing, or does it happen right when you click "Flash"? If it happens in the middle, then this might be indicating a much more alarming issue, caused by a completely different thing.

Bottersnike, 2016-08-23:
I'm not sure quite when it happened as I was away from my pc, but I have flashed my sd card with win32diskimager and checked it with gparted and can confirm that the problem is with Etcher as w32di worked perfectly without any errors.

harlev, 2016-09-12:
I have the same issue on Windows 7. It happens after it downloads the image and almost as soon as it starts writing it
[image]

jviotti, 2016-09-23:
I believe I managed to finally fix this issue. Consult the following PR for details: https://github.com/resin-io-modules/etcher-image-write/pull/64. I'll be closing this issue once the referenced PR is merged, but please don't hesitate in re-opening if you still encounter the error on the next Etcher version.

closed 2016-09-26 by jviotti

wsloand, 2016-09-28:
I just successfully wrote an image to a flash drive with beta.15.  There appeared to be a large number of parameters affecting success, but at least in this case on the same computer that was having trouble before, the write worked!