balena-io/etcher #629: Not enough space error

richard-kimber, 2016-08-07:
- **Etcher ver:1.0.0-beta.13:**
- **Operating system and architecture: Ubuntu 16.04 LTS**
- **Do you see any meaningful error information on DevTools?**
I didn't know about this until afterwards, so no.

I plugged my Raspberry Pi (Max2Play) 16GB Sandisk SD card into my reader and made a copy using:
sudo dd bs=4M if=/dev/sdb | gzip > rasbian.img.gz
I then used Etcher to write this image to another SD card (identical in capacity and make), but eventually got an error saying there was not enough room and that I should try again with a larger card.

However, the process was successful if I used:
gunzip --stdout rasbian.img.gz | sudo dd bs=4M of=/dev/sdb
from the command line:
0+269514 records in
0+269514 records out
15523119104 bytes (16 GB, 14 GiB) copied, 1081.82 s, 14.3 MB/s

Do SD cards need some preparation before using Etcher? - E.g. deleting all partitions?

jviotti, 2016-08-07:
Hi @richard-kimber ,

Thanks a lot for reporting this, which is very weird. If the drive is really not large enough, Etcher would report that and prevent you from writing in the first place (unless you're writing a bz2 image, which we can't know the uncompressed size in advance).

Some thought:

- We might be failing to get the correct drive size, but somehow the size validation gets bypassed for some reason (worth investigating if its the case). In order to discard this hypothesis, can you show me the output of `lsblk -b -d $device --output SIZE`, where `$device` is the device in question? Can you also run this small bash script (when the problematic drive is plugged in) and show me the output: https://github.com/resin-io-modules/drivelist/blob/master/scripts/linux.sh?

- Another hypothesis would be that we're interpreting the estimated uncompressed size from the `gz` image REALLY badly. Can you try decompressing the `.img.gz` and flashing the `.img` directly? Do you get the same error? Also, do you have a link to the *exact* file you're trying to flash?

richard-kimber, 2016-08-08:
lsblk (using /dev/sdk) gives:
       SIZE
15523119104

the script gives:-
device: /dev/sda
description: "Samsung SSD 850 "
size: 128035676160
mountpoint: /boot/efi,/mnt/common,
name: /dev/sda
protected: False
system: True

device: /dev/sdb
description: "Samsung SSD 850 "
size: 256060514304
mountpoint: /,/usr/local,/mnt/scratch,/mnt/record,/media/info,
name: /dev/sdb
protected: False
system: True

device: /dev/sdc
description: "WDC WD10EZRX-00L"
size: 1000204886016
mountpoint: /mnt/backups,/mnt/dat,/media/spare,
name: /dev/sdc
protected: False
system: True

device: /dev/sdd
description: "WDC WD30EFRX-68E"
size: 3000592982016
mountpoint: /media/audio,
name: /dev/sdd
protected: False
system: True

device: /dev/sde
description: "Zalman SSD0060F1"
size: 60022480896
mountpoint: /home,
name: /dev/sde
protected: False                                                                              
system: True                                                                                  
                                                                                              
device: /dev/sdf                                                                              
description: "2105            "                                                               
size: 2000398934016                                                                           
mountpoint:                                                                                   
name: /dev/sdf                                                                                
protected: False                                                                              
system: True                                                                                  
                                                                                              
device: /dev/sdg                                                                              
description: "EZRX-22D8PB0    "                                                               
size: 2000398934016                                                                           
mountpoint:                                                                                   
name: /dev/sdg                                                                                
protected: False                                                                              
system: False                                                                                 
                                                                                              
device: /dev/sdk                                                                              
description: "Multi-Reader  -3"                                                               
size: 15523119104                                                                             
mountpoint: /media/psrwebs/cea0b7ae-2fbc-4f01-8884-3cb5884c8bb7,/media/psrwebs/boot,          
name: /dev/sdk                                                                                
protected: False                                                                              
system: False

I decompressed the image and got the same 'not enough space' error with the full-sized .img file. Maybe there's something funny about the image.  Initially, when I started using Max2Play, I used Etcher successfully to write the (much smaller) image.  I then used Max2Play's facility to extend the filesystem to the full size of the card so that there was room to install other things. It is this 15.5GB image that is causing the problems. Though, as I said, it works OK from the command line with dd.

I'll post a link to the compressed image in a few hours - my upload speed is very slow and it's 1.4GB in size

jviotti, 2016-08-08:
Hi @richard-kimber ,

Thanks a lot for the information. Everything looks fine on that side. I'll wait for the image link to check if I can reproduce myself, and if I can detect anything weird by inspecting the file myself.

richard-kimber, 2016-08-09:
Here is the link:
[https://www.dropbox.com/s/a5712r4mi0tppby/rasbian.img.gz?dl=0](url)

imrehg, 2016-08-19:
@richard-kimber thanks for the test image, now have some idea what's happening, at least in part (@jviotti)

Apparently the gzip file format cannot accurately return the size of files above 4GB (2^32 bytes), but returns the modulo:

That matches the things seen above.
Gzip's own assessment:
```
$ gzip -l rasbian.img.gz 
         compressed        uncompressed  ratio uncompressed_name
         1497486560          2638217216  43.2% rasbian.img
```
which is the same as etcher sees:
```
({"path":"/home/greg/Downloads/rasbian.img.gz","size":2638217216})
```
but the correct file size 15523119104, and then indeed `15523119104 mod 2^32 = 2638217216`.

But this is probably just part of the story, and likely not the source of the failure, if the uncompressed image does not work either. 

I could reproduce the issue by taking an SD, dd-ing it off to a file, and then trying to flash it back with etcher. It goes to 100% then fails with a not enough space error.

```
{"command":"progress","data":{"type":"write","percentage":99,"eta":0,"speed":17603788}}
{"command":"progress","data":{"type":"write","percentage":100,"eta":0,"speed":17611442}}
{"command":"error","data":{"message":"Not enough space on the drive","code":"ENOSPC"}}
```
Though it looks like the contents are written properly (testin: make a copy of the card with dd, clear the partitions & save, flash it with etcher, get not enough space error, dd off again and compare copy: matches)

richard-kimber, 2016-08-19:
Thanks.  So, I should just ignore the error, and unmount the SD and use it normally?

jviotti, 2016-08-19:
Thanks a lot for the valuable investigation :)

> Apparently the gzip file format cannot accurately return the size of
> files above 4GB (2^32 bytes), but returns the modulo:

Right, I read this somewhere :/ Do you know if we have any alternatives?

> Though it looks like the contents are written properly (testin: make a
> copy of the card with dd, clear the partitions & save, flash it with
> etcher, get not enough space error, dd off again and compare copy:
> matches)

This is weird. First, if the estimated size is smaller than the real
size, we shouldn't have hit 100%, right? The `ENOSPC` should have been
thrown before. Second, if the image was not written completely, how is
it possible that you can read back with `dd` and it matches the whole
image?

Can you try filling the drive with `/dev/random` and then flash with
Etcher? Does it still match?

imrehg, 2016-08-19:
@jviotti  

1) **gzip size problem**
Looks like on the command line people recommended something like `zcat file.gz | wc -c` or `gzip -dc test1.img.gz | wc -c` which give the correct value - though then decompresses the the file twice. Might have to do that for gzip in the end, though, since likely >4GB files are common for Etcher's use case. I do think this should be a separate issue, will file it as such.

2) **not enough space**
Sorry, looks like I wasn't clear enough, mixing up these two issues.
The "not enough space" error happens regardless of gzip, whenever the flashed image size is close to the drive's capacity (close to or equal).

I had a found a ~512 MB SD card that I could play with quicker, here's what I've found:

* Drive size: 507379712 B = 495488 KiB = 483.875 MiB
* Images up to 505413632 B = 493568 KiB = 482.0 MiB are burned fine (note, smaller than the drive size)
* Images above that size (but smaller than the drive's size) failed in the end (482 < x < 483.975): *this is the bad behaviour*
* Images above the drive's size are not getting selected, correctly.

@richard-kimber looking at this, I think it's not okay to ignore the error, the last 1-2MB on the drive don't seem to be written correctly (checked with zeroing drive out with `/dev/zero` and burning the image again, the checksum does not match).

richard-kimber, 2016-08-19:
Thanks.

I've looked at the manual and 
zcat file.gz | wc -c

will give the correct size.

jviotti, 2016-08-21:
Hi @imrehg 

> since likely >4GB files are common for Etcher's use case. I do think this should be a separate issue, will file it as such.

Sounds good, let's continue discussion for that particular issue separately.

> Sorry, looks like I wasn't clear enough, mixing up these two issues.
The "not enough space" error happens regardless of gzip, whenever the flashed image size is close to the drive's capacity (close to or equal).

Etcher doesn't write the image byte by byte, but block by block. While the image is indeed a bit smaller than the drive's capacity, the block size (which is usually ~1MB) exceeds the capacity of the drive, and thus throws the error.

I guess we can try to handle this edge case (where the last block causes `ENOSPC`) by splitting the block.

imrehg, 2016-08-22:
Hey @jviotti I did find that it writes by [chunks](https://github.com/resin-io-modules/etcher-image-write/blob/master/lib/write.js#L39), that's why I tried make the test image shorter by a whole chunck, but found that that's not enough, I had to be shorter by almost 2 chuncks to be able to write.

* Writing 482.0 MiB = 482 chunks writes properly
* Above that, e.g. writing 482.1 MiB ~ 483 chunks does runs out of space
* Drive capacity: 483.875 MiB  (so even if in your assessment up to 483.0 MiB = 483 chunk it should be okay, that's not appear to be the case)

Thus I think there's something more than an issue with the last chunk.

jviotti, 2016-08-22:
I see. Looks like we have an off-by-one bug in the conditional that
checks if we can still push one more chunk or not. I'll definitely
investigate and fix.

jviotti, 2016-08-24:
I've sent a PR fixing the off-by-one issue on the last block. I'm intentionally not covering the edge case where you write, e.g: 483.1 MiB to a 483.875 MiB, since we explicitly pad unaligned blocks to our desired block size (1 MiB) on the last block (expanding 483.1 MiB to 484 MiB), otherwise Windows throws EPERM.

jviotti, 2016-08-24:
The PR I sent closes this issue. Let's continue discussion about the particular gz modulo uncompressed size problem in the newly created issue.

closed 2016-08-24 by jviotti