balena-io/etcher #630: Linux Dependencies

a-m-s, 2016-08-08:
- beta 13
- Linux 32/64

Etcher has quite a lot of dependencies, but they're not documented anywhere.

I see at least GTK2, ATK, Gconf, X11, etc, but it's hard to tell because setting `LD_DEBUG=files` causes my entire desktop session to shutdown abruptly (!?)

If the libraries can't be bundled into the AppImage then the list of dependencies (libraries, and/or packages) should be documented somewhere obvious, please.

Or, it might be nice to have the wrapper script check the dependencies and pop up a zenity dialog if a library is missing (beware 32/64 bit issues, however).

jviotti, 2016-08-09:
Hi @a-m-s ,

Thanks a lot for the suggestion. We've been discussing this and we will:

- Document the dependencies in the user documentation.
- Create `*.deb`/`*.rpm` packages and declare dependencies there as well.

a-m-s, 2016-08-09:
Just FYI, there's already an Arch Linux package here:

https://aur.archlinux.org/packages/etcher/

It's just a wrapper for the AppImage, but it'd be nice if it were a native binary.

jviotti, 2016-08-24:
I've just sent a PR adding explicit documentation about all required runtime dependencies. We also have another opened GitHub issue to properly package Etcher for mainstream Linux distributions, declaring dependencies as necessary (see https://github.com/resin-io/etcher/issues/632).

Thanks a lot!

closed 2016-08-24 by jviotti