balena-io/etcher #631: not working !

amir1376irani, 2016-08-08:
- **Etcher version :  1.0.0-beta.13 **
- **Operating system and architecture : Ubuntu 16.04 LTS 64 bit**

Iso file and flash drive is selected and after hitting " Flash! " it ask my root password, i enter it but after 3 sec everything goes back to what it was !! no change happens , i tried more than 10 times but it's crazy unusable, a really need help... by the way thank from support

imrehg, 2016-08-08:
Hi, can you start Etcher, open DevTools by pressing `Ctrl+Alt+I`, try burning the image like you tried,, and sending us if there's any error message in that DevTools console?

istorrs, 2016-08-29:
I am seeing the same issue, which is present on beta 13, but not on beta 12 or earlier. After entering the root password, no flashing takes place. The UI returns from "Starting" to Flash.

However, if I run etcher as sudo sudo ./Etcher-linux-x64.AppImage, the burning works.

jviotti, 2016-08-30:
Hi @istorrs ,

I wonder what could be causing this. There is clearly something going on with the writer elevation in some systems. What GNU/Linux distribution and version are you running? 

When you click "Flash" (without running the app as `sudo`), you get the elevation dialog (based on `pkexec` or `kdesudo` depending on your system). If you expand the details of such dialogs, you can see the actual full command being ran. 

*Without putting your password in the elevation dialog nor closing it (otherwise certain temporary files referred by the command could be deleted)*, can you try running that command in another terminal? What do you get?

istorrs, 2016-08-30:
I am using Ubuntu 16.04 with latest updates applied daily.

When I click "Flash" I get a dialog with title "Authenticate" asking for my password.

The details show the following:
Action: org.freedesktop.policykit.exec
Vendor:  http://hal.freedesktop.org/docs/PolicyKit/

If I enter a bogus password to this dialog, the dialog vibrates and reports that my authentication attempt was unsuccessful.

If I enter the correct password, the dialog disappears, and the etcher button reverts from "Starting" back to "Flash" exactly as if I had pushed cancel on the dialog.

jviotti, 2016-08-31:
Hm, so you don't get the command at all (that's weird, `pkexec` should
be showing it)? Can you check DevTools (by pressing Ctrl+Alt+I)? The
actual command being ran might have been printed there.

dstinebaugh, 2016-09-17:
Same boat here, Linux Mint 18 x64. running with sudo works fine in beta 14, however just sit's at "starting" in the ui. dev tools just report that it's waiting for elevation, didn't actually see any errors. happy to share any logs needed.

jviotti, 2016-09-20:
Hi  @dstinebaugh ,

Thanks for reaching out! I suggest trying the following, which may reveal the cause of the error:

- [As mentioned above](https://github.com/resin-io/etcher/issues/631#issuecomment-243767259), can you check what is the exact command being elevated from either the `pkexec` or `kdesudo` prompt? (you can find in the dialog details, which can be expanded to reveal the command).
- *Keep the password prompt opened* (don't cancel it nor submit it, since the command references temporal feels that will be removed if you close the dialog), and run the command you found before in a terminal, and see what you get.

Hopefully that reveals the origin of the issue. Let me know!

jviotti, 2016-09-26:
I'm assuming this issue is related to https://github.com/resin-io/etcher/issues/665, which describes the exact same problem caused when the file name contains a space.

I have a fix for this issue, which will land in the next release. Please don't hesitate in re-opening if you find out this issue represents another problem!

Bunker, 2016-10-04:
Ok, together with @jviotti on on the gitter chat I figured out that the problem has to do with spaces in the file name of the AppImage. 

See c/p from our gitter conversation below:

> when there is already a etcher. AppImage in the same directory, the new download gets the name Etcher-linux-x64 (1).AppImage once the permissions are set it runs as it is supposed to be except for the flashing part doesn't work
> when I rename the file to Etcher-linux-x64.AppImage the flashing works as expected.
> I tested this again by renaming the AppImage again and adding " test" and the flashing fails again.
> so it has to do with the space in the file name.

Also here are the contents of the array in the development console:

`"/tmp/.mount_6n4tEJ/usr/bin/resources/app.asar"
1:"/home/[username]/Downloads/Manjaro-Arm_rpi2_base_1608.zip"
2:"--robot"
3:"--drive"
4:"/dev/mmcblk0"
5:"--unmount"
6:"--check"
7:"--log"
8:"/tmp/etcher-2969edxw9VmpsL7J.log"`

jviotti, 2016-10-27:
This issue has been fixed by https://github.com/resin-io/etcher/pull/779

closed 2016-10-27 by jviotti

lurch, 2016-10-27:
Double-quotes FTW! :laughing: