balena-io/etcher #636: Windows 10 error unless run as administrator

zacchaeusluke, 2016-08-17:
- **Etcher version:** v1.0.0-beta.13
- **Operating system and architecture:** Windows 10 1607 AMD x64
- **Do you see any meaningful error information on DevTools?** This error text below:
```
tracker.js:10 WRITER: Attempting to elevate
tracker.js:10 Wed Aug 17 2016 17:43:30 GMT+0100 (GMT Summer Time) Flash error
tracker.js:10 Error: { Error: Command failed: C:\Users\zacchaeus\AppData\Local\Temp\8AF3.tmp.exe -c -n -h -w set ELECTRON_RUN_AS_NODE=1 && set ETCHER_INTERNAL_LOG_FILE=C:\Users\ZACCHA~1\AppData\Local\Temp\etcher-10176ErEmPClzcjQr.log && call C:\Program Files (x86)\Etcher\Etcher.exe C:\Program Files (x86)\Etcher\resources\app.asar\lib\src\child-writer\writer-proxy.js C:\Program Files (x86)\Etcher\resources\app.asar C:\Users\zacchaeus\Desktop\Raspberry pi\2016-05-27-raspbian-jessie.img --robot --drive \\.\PHYSICALDRIVE1 --unmount --check --log C:\Users\ZACCHA~1\AppData\Local\Temp\etcher-10176ErEmPClzcjQr.log

    at ChildProcess.exithandler (child_process.js:214:12)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at maybeClose (internal/child_process.js:850:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:215:5)
  cause: 
   { Error: Command failed: C:\Users\zacchaeus\AppData\Local\Temp\8AF3.tmp.exe -c -n -h -w set ELECTRON_RUN_AS_NODE=1 && set ETCHER_INTERNAL_LOG_FILE=C:\Users\ZACCHA~1\AppData\Local\Temp\etcher-10176ErEmPClzcjQr.log && call C:\Program Files (x86)\Etcher\Etcher.exe C:\Program Files (x86)\Etcher\resources\app.asar\lib\src\child-writer\writer-proxy.js C:\Program Files (x86)\Etcher\resources\app.asar C:\Users\zacchaeus\Desktop\Raspberry pi\2016-05-27-raspbian-jessie.img --robot --drive \\.\PHYSICALDRIVE1 --unmount --check --log C:\Users\ZACCHA~1\AppData\Local\Temp\etcher-10176ErEmPClzcjQr.log
   
       at ChildProcess.exithandler (child_process.js:214:12)
       at emitTwo (events.js:106:13)
       at ChildProcess.emit (events.js:191:7)
       at maybeClose (internal/child_process.js:850:16)
       at Process.ChildProcess._handle.onexit (internal/child_process.js:215:5)
     killed: false,
     code: 3221225781,
     signal: null,
     cmd: 'C:\\Users\\zacchaeus\\AppData\\Local\\Temp\\8AF3.tmp.exe -c -n -h -w set ELECTRON_RUN_AS_NODE=1 && set ETCHER_INTERNAL_LOG_FILE=C:\\Users\\ZACCHA~1\\AppData\\Local\\Temp\\etcher-10176ErEmPClzcjQr.log && call C:\\Program Files (x86)\\Etcher\\Etcher.exe C:\\Program Files (x86)\\Etcher\\resources\\app.asar\\lib\\src\\child-writer\\writer-proxy.js C:\\Program Files (x86)\\Etcher\\resources\\app.asar C:\\Users\\zacchaeus\\Desktop\\Raspberry pi\\2016-05-27-raspbian-jessie.img --robot --drive \\\\.\\PHYSICALDRIVE1 --unmount --check --log C:\\Users\\ZACCHA~1\\AppData\\Local\\Temp\\etcher-10176ErEmPClzcjQr.log' },
  isOperational: true,
  killed: false,
  code: 3221225781,
  signal: null,
  cmd: 'C:\\Users\\zacchaeus\\AppData\\Local\\Temp\\8AF3.tmp.exe -c -n -h -w set ELECTRON_RUN_AS_NODE=1 && set ETCHER_INTERNAL_LOG_FILE=C:\\Users\\ZACCHA~1\\AppData\\Local\\Temp\\etcher-10176ErEmPClzcjQr.log && call C:\\Program Files (x86)\\Etcher\\Etcher.exe C:\\Program Files (x86)\\Etcher\\resources\\app.asar\\lib\\src\\child-writer\\writer-proxy.js C:\\Program Files (x86)\\Etcher\\resources\\app.asar C:\\Users\\zacchaeus\\Desktop\\Raspberry pi\\2016-05-27-raspbian-jessie.img --robot --drive \\\\.\\PHYSICALDRIVE1 --unmount --check --log C:\\Users\\ZACCHA~1\\AppData\\Local\\Temp\\etcher-10176ErEmPClzcjQr.log' }

    at Socket.child.stderr.on (C:\Program Files (x86)\Etcher\resources\app.asar\lib\src\child-writer\index.js:85:29)
    at emitOne (events.js:96:13)
    at Socket.emit (events.js:188:7)
    at readableAddChunk (_stream_readable.js:172:18)
    at Socket.Readable.push (_stream_readable.js:130:10)
    at Pipe.onread (net.js:542:20)
```

[image]

Then when I press ok a banner says 'Your removable drive may be corrupted'. I'm trying to flash a new sd card which I've formatted, and it worked fine flashing with USB Image Tool.

I ran etcher with admin privileges and it now works, so this appears to be some problem with elevating privileges.

jviotti, 2016-08-17:
Hey there,

Thanks a lot for reporting. This is indeed a very weird one. Can you check if you can reproduce back if running as a normal user and going through the UAC elevation dialog again?

If so, can you run the command being reported as failing (`C:\Users\zacchaeus\AppData\Local\Temp\8AF3.tmp.exe -c -n -h -w set ELECTRON_RUN_AS_NODE=1 && set ETCHER_INTERNAL_LOG_FILE=C:\Users\ZACCHA~1\AppData\Local\Temp\etcher-10176ErEmPClzcjQr.log && call C:\Program Files (x86)\Etcher\Etcher.exe C:\Program Files (x86)\Etcher\resources\app.asar\lib\src\child-writer\writer-proxy.js C:\Program Files (x86)\Etcher\resources\app.asar C:\Users\zacchaeus\Desktop\Raspberry pi\2016-05-27-raspbian-jessie.img --robot --drive \\.\PHYSICALDRIVE1 --unmount --check --log C:\Users\ZACCHA~1\AppData\Local\Temp\etcher-10176ErEmPClzcjQr.log` in this case, but will be slightly different for sure, **while you keep Etcher and the error window open** (otherwise the temp files are wiped out)? What do you get?

zacchaeusluke, 2016-08-18:
The UAC elevation dialog never appears.

When trying to run that command I get a system Error:
The program can't start because MSVCR120.dll is missing from your computer. Try reinstalling the program to fix this problem.

I repaired the Visual C++ Redistributable Packages for Visual Studio 2013 (which didn't seem to do anything)
Then I uninstalled/reinstalled Etcher and it now seems to work.

jviotti, 2016-08-18:
Hi @zacchaeusluke ,

Thanks for the quick answer. As far as I can see, that file is a library that should be coming from Visual Studio 2013. My bet is that repairing Visual Studio 2013 fixed the issue, but took an Etcher re-install for the program to be re-configured correctly.

I'm closing this as resolved, but please re-open if the issue ever happens again!

closed 2016-08-18 by jviotti