balena-io/etcher #637: Etcher leaves behind process after burning and exiting the UI

imrehg, 2016-08-19:
- Etcher 1.0.0-beta13
- Arch Linux 64bit

Starting Etcher and burning a file, afterwards exiting leaves an etcher process behind. It does not happen if the program is started, not burned, and exited. I guess this means something is up with the elevation?

Here's my system after 3 start-burn-and-exit:

```
$ps uax |grep etcher
greg      1376  0.2  1.5 974088 127336 pts/9   Sl   11:56   0:01 /tmp/.mount_zADdyB/usr/bin/etcher /tmp/.mount_zADdyB/usr/bin/resources/app.asar/lib/src/child-writer/writer-proxy.js /tmp/.mount_zADdyB/usr/bin/resources/app.asar /home/greg/Downloads/FreeBSD-10.3-RELEASE-arm-armv6-RPI-B.img.xz --robot --drive /dev/sdf --unmount --check --log /tmp/etcher-563Ws0zjh0K4YSQ.log
greg      9482  0.0  1.5 965892 125976 pts/3   Sl   10:35   0:02 /tmp/.mount_8MehiU/usr/bin/etcher /tmp/.mount_8MehiU/usr/bin/resources/app.asar/lib/src/child-writer/writer-proxy.js /tmp/.mount_8MehiU/usr/bin/resources/app.asar /home/greg/Downloads/ada/2016-03-25-28c.img.xz --robot --drive /dev/sdf --unmount --check --log /tmp/etcher-8512x2nVk1SDGXhM.log
greg     21982  0.0  1.5 958720 126156 pts/7   Sl   11:26   0:01 /tmp/.mount_3avsg6/usr/bin/etcher /tmp/.mount_3avsg6/usr/bin/resources/app.asar/lib/src/child-writer/writer-proxy.js /tmp/.mount_3avsg6/usr/bin/resources/app.asar /home/greg/Downloads/resin-RapiroRPi3-1.8.0-1.13.0-7ad55d1c946b.img --robot --drive /dev/sdf --unmount --check --log /tmp/etcher-212983245IIEEg1b0.log
```

jviotti, 2016-08-19:
Hi @imrehg ,

Yeah, the problem is that is quite hard to ensure previously used custom mount points get unmounted. AppImages solve this issue by using FUSE, which ensures the AppImage is automatically unmounted when is no longer used, but FUSE prevented us from implementing elevation, so we're doing a basic mounting that is not guaranteed to go away.

Some alternatives could be:

- When the AppImages is mounted, see if it was previously mounted and re-use that. This somewhat mitigates the issue, but there might always be one remaining mount point anyway.

- Find a way to ensure that the AppImage gets unmounted.

- Find a way to implement elevation some other way, without re-mounting stuff.

imrehg, 2016-08-22:
Yeah, it does look like a tricky topic. Don't know enough about the issue to comment knowledgeably, but the 2nd option seems to be the most tempting...

probonopd, 2016-08-25:
The AppImage is supposed to unmount after AppRun exits.

jviotti, 2016-08-25:
Hi @probonopd,

Right, but remember we're doing some kind of a hack to be able to run
the contents of the AppImage as `root` (see
https://github.com/probonopd/AppImageKit/issues/146).

As a recap, once the AppImage is mounted with FUSE, and the user accepts
elevation, we re-mount the AppImage without FUSE. There seems to be some
edge case where our `umount` routine doesn't get called.

jviotti, 2016-10-25:
So currently we're re-mounting the AppImage without FUSE in order to execute the CLI as `root`. Since we have `root` permissions at this point, why not remounting the AppImage as `root` instead? This means that only `root` will have access to the re-mounted image, and that FUSE will make sure the image gets unmounted automatically.

closed 2016-10-26 by jviotti