balena-io/etcher #639: Status bar goes beyond 100% with my .zip image

rswamy, 2016-08-19:
- **Etcher version:** 1.0.0-beta.13
- **Operating system and architecture:** Mac OSX 10.11.6 (15G31)
- **Do you see any meaningful error information on DevTools?** 

No, there are no errors I see in the console. Looks like it is stuck at 99% but the loading bar continues on.

[image]

[image]

My image is a .zip file. When I unzipped it with the default Archive Utility I got a .cpgz file which indicates the file may be corrupted or wasn't unzipped correctly. I tried unzipping in terminal with the unzip command but got this error:

```
Archive:  udoo_quad_revC_yocto_090713.zip
warning [udoo_quad_revC_yocto_090713.zip]:  76 extra bytes at beginning or within zipfile
  (attempting to process anyway)
error [udoo_quad_revC_yocto_090713.zip]:  reported length of central directory is
  -76 bytes too long (Atari STZip zipfile?  J.H.Holm ZIPSPLIT 1.1
  zipfile?).  Compensating...
   skipping: udoo_quad_revC_yocto_09072013.img  need PK compat. v4.5 (can do v2.1)

note:  didn't find end-of-central-dir signature at end of central dir.
  (please check that you have transferred or created the zipfile in the
  appropriate BINARY mode and that you have compiled UnZip properly)
```

However, unzipping the file with The Unarchiver created a .img file which burned successfully.

So I think there either may be an issue with the way etcher is unzipping a file before flashing it (are you using the native unzip command?) or an issue with the way my image was zipped up. For now to be 'safe' I will manually unzip image files before providing them to etcher.

jviotti, 2016-08-21:
Hey there,

Thanks a lot for reporting. Looks like the zip is indeed very corrupted, but somehow The Unarchiver manages to deal with it correctly. 

Can you link me to the exact zip you're trying to flash? I'll try to see if there is something we can do on our end to either manage to interpret the zip correctly, or show an appropriate error message instead of breaking the UI so badly.

jviotti, 2016-08-24:
I managed to find the exact zip file you were using, and can confirm the bug was fixed by https://github.com/resin-io-modules/etcher-image-stream/pull/27, which will land in the next release.  The problem was that the zip library we were using to scan the entries returned an incorrect uncompressed size for certain zips, including yours.

Please re-open if you still encounter any issues in the next release! Thanks a lot!

closed 2016-08-24 by jviotti