balena-io/etcher #640: Can't select drive

dubcl, 2016-08-20:
- **Etcher version:** 1.0.0-beta13 x64
- **Operating system and architecture:** Debian Stretch amd64
- **Do you see any meaningful error information on DevTools?** no

Hi, i try to use etcher with a volumio img, but when i can't select the drive and the app don't show any info or new window to select drive, the devtool only show

```
Sat Aug 20 2016 00:26:46 GMT-0300 (CLST) Application start
tracker.js:10 Sat Aug 20 2016 00:26:46 GMT-0300 (CLST) Checking for updates
tracker.js:10 Sat Aug 20 2016 00:26:56 GMT-0300 (CLST) Select image ({"path":"/media/carlos/storage/Isos/volumio-0.976-2016-07-28-x86.img","size":3670016000})
```
When the app start show a window with:
```
/tmp/.org.chromium.Chromium.976xn8: line 12: blkid: command not found
/tmp/.org.chromium.Chromium.976xn8: line 12: blkid: command not found
```
But blkid is already installed, fuse too.  

And when I click on "connect a drive" the app don't do nothing, I try to use a usb stick and a hard drive to burn the image.

Cheers.

gameslayer, 2016-08-21:
Confirmed same issue, I regret updating it as I really need it right now..

jviotti, 2016-08-21:
Hey there,

Thanks for reporting, and sorry to hear you're having problems. Nothing has been changed in the packaging and drive detection side that could be suddenly triggering this issue.

As far as I can see, the error comes from [drivelist](https://github.com/resin-io-modules/drivelist), the module we use to scan available drives (that calls a bash script), so it makes sense that Etcher refuses to let you select any drive.

Can you try to manually run this script and report back if you get any errors? https://github.com/resin-io-modules/drivelist/blob/master/scripts/linux.sh

Also, you can find the previous version (v1.0.0-beta.12) here: https://resin-production-downloads.s3.amazonaws.com/etcher/1.0.0-beta.12/Etcher-linux-x64.AppImage. Let me know if somehow running the older version fixes the issue.

jviotti, 2016-08-24:
Hi @dubcl , @gameslayer ,

- Can you check where is `blkid` located? Is such path present in the PATH environment variable? 
- Could it be that its added to your shell, but its not present when running the AppImage directly?
- Can you try manually running the AppImage from the terminal (`./Etcher.AppImage`)? Does it work there?

jviotti, 2016-08-24:
I was able to reproduce with a cleanly installed Debian Stretch. The issue is indeed that `blkid` is not in the `PATH` (at least not on Debian by default). I have it in `/sbin/blkid`. Adding `/sbin` to the `PATH` before running Etcher works fine for me, however we should either find the correct location before bailing out, or use another tool to get the mount points.

closed 2016-08-26 by jviotti