balena-io/etcher #642: Windows 10 Unexpected end of json

Bottersnike, 2016-08-22:
- Etcher version 1.0.0-beta.13
- Windows 10 64bit
- DevTools printed the following:
tracker.js:10 Error: undefined:1
{

SyntaxError: Unexpected end of JSON input
    at Object.parse (native)
    at EventEmitter.tail.on (C:\Program Files (x86)\Etcher\resources\app.asar\lib\src\child-writer\writer-proxy.js:76:25)
    at emitOne (events.js:96:13)
    at EventEmitter.emit (events.js:188:7)
    at ReadStream.<anonymous> (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\file-tail\index.js:52:19)
    at emitOne (events.js:96:13)
    at ReadStream.emit (events.js:188:7)
    at readableAddChunk (_stream_readable.js:172:18)
    at ReadStream.Readable.push (_stream_readable.js:130:10)
    at onread (fs.js:1743:12)
    at FSReqWrap.wrapper [as oncomplete] (fs.js:675:17)

    at Socket.child.stderr.on (C:\Program Files (x86)\Etcher\resources\app.asar\lib\src\child-writer\index.js:85:29)
    at emitOne (events.js:96:13)
    at Socket.emit (events.js:188:7)
    at readableAddChunk (_stream_readable.js:172:18)
    at Socket.Readable.push (_stream_readable.js:130:10)
    at Pipe.onread (net.js:542:20)

It stopped at 45%. I'm retrying now and will edit this with the result of a retry.

jviotti, 2016-08-22:
Hey there,

Thanks a lot for reporting. This is an interesting issue. This is what I
believe its happening:

The Etcher GUI basically spawns the Etcher CLI as a child process that
emits JSON lines that are then read and interpreted by the GUI. Each
line is supposed to contains a fully valid JSON object, but that seems
to fail in your setup for some reason (the buffer gets full, or the data
is flushed before the whole line has been received for some reason).

Is this something you can actively reproduce?

If so, you can see that we print the path to a temporary file in
DevTools (which you can open by pressing `Ctrl+Alt+I`) right when you
start a flash, which is the location where the JSON lines are temporary
saved.

Can you start a flash, wait until it fails with this error, and without
closing Etcher (otherwise the temp file will be wiped out), send us what
the temporary file contains?

I'm wondering if your system is emitting a huge speed number, or
something else that could trigger this.

Bottersnike, 2016-08-22:
I forget to make the edit, but when I retried, it all worked fine. I'll see if I can replicate it.

jviotti, 2016-08-22:
Yeah, I had the impression this might be not be easy to reproduce. Let's
keep an eye on it. I'm going to be flashing some stuff on Windows 10 and
see if I get anything similar, and will also be researching ways to
ensure the whole line is written to the temporary file.

jviotti, 2016-09-20:
A quick solution to this problem is to ignore malformed JSON lines (they are purely for state information, so ignoring one should make no harm, since the state will be updated with the next one), but I'm still trying to find the real source of the problem.

jviotti, 2016-09-20:
Hm, after some logging and hundreds of flashes attempting to reproduce, it seems that the CLI sends a line containing a single open curly brace:

```
{
```

Which makes no sense at all :/ 

Still investigating...

informatik-tutorial, 2016-10-04:
the issue seems to be still unresolved.
Here is a log file where the error shows up several times.

[-1475589401095.log.txt](https://github.com/resin-io/etcher/files/508646/-1475589401095.log.txt)

bkize, 2016-10-15:
This issue just occurred to me. It happened during the image validation process at 99% complete. The image was written successfully. Running on a very current build of Windows 10 64-bit - v1607-b14931. Etcher is v1.0.0-beta.15. Willing to help repro, test, etc. with some guidance. I couldn't find a log file but here's a screen snap of the failure:

[image]

lurch, 2016-10-18:
Looks like it might be some kind of output buffering / flushing problem?

WasabiFan, 2016-10-18:
I'd guess that @lurch is right with the diagnosis, however there definitely could be other things causing it. I'd suggest, with the nature of the JSON you're parsing, that you look into a streaming parser; that way you can just spew JSON and it will be parsed as it arrives.

closed 2016-10-26 by jviotti