balena-io/etcher #643: Show an MD5 checksum along with the CRC32 one

jviotti, 2016-08-22:
From https://github.com/resin-io/etcher/issues/357#issuecomment-241286904

jviotti, 2016-08-23:
@taahirisaacs Can you give me a hand with how we should display this? We already show the CRC32 checksum, as shown in the screenshot, but we also want to show the MD5.

[image]

jviotti, 2016-08-23:
Also, the MD5 would probably be much larger. Something like `af39449417abe3e688f3808566f9a76e`.

jviotti, 2016-08-23:
Alternatively, we could provide an option in the settings page so the user can choose which one to display (CRC32 or MD5), but I guess its much more preferable to have both by default.

alexandrosm, 2016-08-23:
how about a "show md5" under the crc checksum that launches a modal?

jviotti, 2016-08-23:
Could be. Given that many people already suggested MD5, and no-one seems
to like our CRC32 choice (given MD5 seems to be mainstream for this kind
of stuff), what about ditching away CRC32 and just show an MD5 instead?

alexandrosm, 2016-08-23:
sounds good to me

lurch, 2018-02-23:
@jhermsmeier  This can probably be closed, given that the checksum (CRC32 or otherwise) is never actually shown to the user any more.

commit f72cc6bb (2018-03-09): Merge pull request #2108 from resin-io/switch-chksm-algo

fix(perf): Remove support for CRC32 checksumming

Close #643
closed 2018-03-09 by jhermsmeier