balena-io/etcher #649: Etcher cannot recognise 'SanDisk Cruzer Blade' flash drive under OS X.

UniK, 2016-08-23:
- **Etcher version:** 1.0.0-beta.13
- **Operating system and architecture:** OS X 10.11.6; Kernel Version 15.6.0; x86_64
- **Do you see any meaningful error information on DevTools?** No
- **Sytem Console Messages:** (not sure if relevant)
```
- Etcher[43486]: BUG in libdispatch client: kevent[EVFILT_MEMORYSTATUS] add: "Operation not supported" - 0x2d
- Etcher[43486]: assertion failed: 15G31: libxpc.dylib + 78286 [54D1328E-054E-3DAA-89E2-375722F9D18F]: 0x89
```

I have two of this flash drives, and both are not visible for Etcher under OS X.
They are both recognised by Etcher running on Ubuntu 16.04.

jviotti, 2016-08-23:
Hey there,

Thanks a lot for reporting. We use a module called
[drivelist](https://github.com/resin-io-modules/drivelist) to detect the
available drives, so something might be going wrong there.

Can you run the following Bash script while the problematic drive is
plugged in and share the results?
https://github.com/resin-io-modules/drivelist/blob/master/scripts/darwin.sh

The output of the following commands would also be useful:

- `diskutil list`
- `diskutil info /dev/diskN`, where `N` is the id of the problematic
  drive.

UniK, 2016-08-23:
```
$ ./darwinsh.sh
...

device: /dev/disk2
description: Cruzer Blade
size: 4004511744
mountpoint: 
name: /dev/disk2
raw: /dev/rdisk2
protected: False
system: True

...
```

```
$ diskutil list /dev/disk2
/dev/disk2 (external, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:     FDisk_partition_scheme                        *4.0 GB     disk2
   1:                 DOS_FAT_32 LNX                     4.0 GB     disk2s1

```

```
$ diskutil info /dev/disk2
   Device Identifier:        disk2
   Device Node:              /dev/disk2
   Whole:                    Yes
   Part of Whole:            disk2
   Device / Media Name:      Cruzer Blade

   Volume Name:              Not applicable (no file system)

   Mounted:                  Not applicable (no file system)

   File System:              None

   Content (IOContent):      FDisk_partition_scheme
   OS Can Be Installed:      No
   Media Type:               Generic
   Protocol:                 USB
   SMART Status:             Not Supported

   Total Size:               4.0 GB (4004511744 Bytes) (exactly 7821312 512-Byte-Units)
   Volume Free Space:        Not applicable (no file system)
   Device Block Size:        512 Bytes

   Read-Only Media:          No
   Read-Only Volume:         Not applicable (no file system)

   Device Location:          External
   Removable Media:          No

   Virtual:                  No
   OS 9 Drivers:             No
   Low Level Format:         Not supported
```

jviotti, 2016-08-23:
Hey there,

Thanks a lot for the reports. So the issue is that your operating system detects your drive as non being removable, which causes Etcher to omit it for safety reasons (see ` Removable Media:          No` in `diskutil info`).

I don't think there is much we can do from our side (I don't know what to do if your operating system doesn't think its removable). I suggest to either:

- Find a way to be able to tell (or trick) your OS that the drive in question is indeed removable.
- Enable the "unsafe" mode in Etcher (see the cog at the top right corner), so Etcher doesn't filter non-removable drives (but proceed with caution of course).

closed 2016-08-23 by jviotti