balena-io/etcher #650: EPERM: opereration not permitted, write

Bottersnike, 2016-08-23:
- Etcher version 1.0.0-beta.13
- 64 Bit Windows 10
I get the error:
```
tracker.js:10 Error: EPERM: operation not permitted, write
    at ChildProcess.child.on (C:\Program Files (x86)\Etcher\resources\app.asar\lib\src\child-writer\index.js:94:23)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at handleMessage (internal/child_process.js:718:10)
    at Pipe.channel.onread (internal/child_process.js:444:11)
```
flashing to a perfectly okay SD card.
The img is the latest one from https://www.raspberrypi.org/downloads/raspbian/

jviotti, 2016-08-23:
Hey there!

Thanks a lot for reporting. This looks like the same issue as https://github.com/resin-io/etcher/issues/627. I'll close this one so we can continue discussion on the other issue. 

The submitter was able to get more information about the error using the Event System Log. Can you confirm you get the same thing? If so, do you mind trying the `diskpart` suggestion?

closed 2016-08-23 by jviotti

Lulucmy, 2022-05-18:
I'm just leaving this message in case someone else has the same issue.
If your antivirus is Windows Defender (default on Windows) you have to disable [Controlled folder access](https://support.microsoft.com/en-us/windows/allow-an-app-to-access-controlled-folders-b5b6627a-b008-2ca2-7931-7e51e912b034) in order to flash with Etcher.

magiva, 2022-11-04:
> I'm just leaving this message in case someone else has the same issue. If your antivirus is Windows Defender (default on Windows) you have to disable [Controlled folder [...]

YOURE AMAZING

why isnt this in the main thread of balena etcher.
utter life saver