balena-io/etcher #656: Child process exitted with error code: 1

toyota790, 2016-08-24:
- **Etcher version: v1.0.0-beta.13**
- **Operating system: Windows 10 Professional 64 bit**

It shows the error message as below when I burn the resinOS to SD card. So I can't burn the image to the SD card. Does anyone know how to solve ? Thanks!

[image]

- **DevTools printed the following:**

```
tracker.js:10
Error: Child process exitted with error code: 1
    at ChildProcess.child.on (C:\Program Files (x86)\Etcher\resources\app.asar\lib\src\child-writer\index.js:115:38)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at maybeClose (internal/child_process.js:850:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:215:5)a.(anonymous function) @ tracker.js:10
```
```
tracker.js:10
Error: EPERM: operation not permitted, write
    at ChildProcess.child.on (C:\Program Files (x86)\Etcher\resources\app.asar\lib\src\child-writer\index.js:94:23)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at handleMessage (internal/child_process.js:718:10)
    at Pipe.channel.onread (internal/child_process.js:444:11)
```

jviotti, 2016-08-24:
Hi @toyota790 ,

Thanks a lot for reporting. These issues are really hard to debug since Windows seems to like to throw `EPERM` without any explanation on all sort of situations. That being said:

- Can you actively reproduce the issue? Does it go away if you keep retrying?
- Can you test this custom build: https://www.dropbox.com/s/pva5qpo2sg0g0zo/Etcher-win32-x64-Aug-23-2016.exe?dl=0 and report back how it goes? This version contains some fixes to other related Windows issues, so it may work for you.
- Can you try completely erasing your drive and try again? Windows physical drive access policies say that `EPERM` might be thrown if the drive has a file-system. We should be taking care of this automatically, but lets test anyway. You can do this by running `cmd.exe`, typing `diskpart`, checking your available disks with `list disk`, `select disk N` (where N is the corresponding number from the previous command), `clean`, `rescan`.

toyota790, 2016-08-24:
Hi @jviotti ,
Thanks for your quick reply.

* Yes, I tried many times, and it still showed the same error message. However, I restarted the application and burned the image again, it showed up another error message (As below).

[image]

* This version does not work for me, and it showed up the "Child process exitted with error code: 1" error message again.

* I can't erase my SD card by DiskPart. When I typed the "clean" command in DiskPart, then it printed out the "DiskPart has encountered an error encountered: Access is denied.". But I can erase the SD card with my Macbook. So I am sure that the SD card works normally.

jviotti, 2016-08-24:
> * I can't erase my SD card by DiskPart. When I typed the "clean"
> command in DiskPart, then it printed out the "DiskPart has encountered
> an error encountered: Access is denied.". But I can erase the SD card
> with my Macbook. So I am sure that the SD card works normally.

The fact that `diskpart` complained with an access denied error means
that there is something else preventing your SDCard from being written
to in Windows, and therefore the Etcher error you see makes sense.

The tricky part is discovering what is causing your "Access is denied"
error in DiskPart. Do you have any anti-virus or other software that
might be trying to constantly read removable drives in your computer? If
so, can you try disabling them and trying again, and see if that makes
any difference?

The fact that you can erase it just fine with your MacBook means that
there's definitely something in your Windows installation conflicting
with this.

Can you check if something of interest comes up in the System Event Log
right after you get such DiskPart error? See
https://technet.microsoft.com/en-us/library/aa996634(v=exchg.65).aspx

jviotti, 2016-08-24:
I just realised this is a duplicate of https://github.com/resin-io/etcher/issues/627. Please let me know how it goes in that other thread!

closed 2016-08-24 by jviotti