balena-io/etcher #661: Allow customisation of Etcher's colour scheme

jviotti, 2016-08-25:
Potentially from an archive metadata.

jviotti, 2016-08-31:
After some thought, I believe this can be very easily solved with native CSS variables, which as it seems, they are dynamic, and we can change them at runtime, elegantly affecting every CSS component that uses them. See https://philipwalton.com/articles/why-im-excited-about-native-css-variables/. I think we can find a nice way to have SCSS and native CSS variables to operate together so we can introduce this feature without much friction.

As an alternative, we can:

- Completely decouple the "color" rules from the "structure" rules. 
- We can declare a bunch of CSS classes that match the colors we want to allow the user to customise, like `.theme-background-color`, `.theme-buttom-primary-text-color`.
- Make use of the new classes by `@extend`ing them.
- Simply modify the `.theme-*` rules at runtime. The fact that we're using `@extend` on them should make the changes reflect on the components that use them.

In any case, we must solve the color variable mess that we have at the moment, and decide on a sensible list of colors that we want to give users access to, potentially as a layer of abstraction above components.

jviotti, 2016-09-12:
In order for this task to be feasible, the first step is to drastically simplify the user interface and the colors currently being used there. Ideally, we'll have 3 "main" colors, and there will be straightforward rules to infer the remaining ones based on that.

This means:

- Making sure we're not using slightly different colors in more than one location if using just one makes more sense.
- Reducing extra colors as much as possible without sacrificing the experience.

jviotti, 2016-11-08:
See http://web.archive.org/web/20101101160133/http://blog.dabbledb.com/2007/04/white--or-green.html for some crazy inspiration.

The goal would be to have extended archives specify at most 3 primary colors, and build color palettes on the fly that are ensured to look good, drawing ideas from color theory, etc.

lurch, 2016-11-08:
> See http://web.archive.org/web/20101101160133/http://blog.dabbledb.com/2007/04/white--or-green.html for some crazy inspiration.

That's pretty neat! :-D
As long as there's also an option for the users to manually specify their colour scheme too.

jviotti, 2016-11-08:
I guess we could allow users to override certain colors if really needed, but that should be really discouraged though.

WasabiFan, 2016-11-08:
Honestly, I don't think that you would need to allow overrides. If you do, inevitably people are going to utilize the feature, and users will end up with an inconsistent experience with Etcher: First they see a pretty, well-designed app, but then it changes into something much uglier upon image selection. Start with the automatic option and then add a manual override if it is largely requested.

lurch, 2016-11-08:
I was just thinking that an application that got auto-coloured with the bold colours from e.g. the [Raspberry Pi logo](https://www.raspberrypi.org/trademark-rules/) might look a bit too "intense"? ;)
But I'm not a graphic designer, so feel free to ignore me...

WasabiFan, 2016-11-08:
That's definitely something to consider. It would be nice to see how a basic implementation of the proposed algorithm handled this case and then adjust expectations accordingly. Honestly, this is something that I would personally be interested in attempting if I had some free time and wasn't loaded with other GitHub projects to help maintain :laughing: