balena-io/etcher #665: Spaces in filename breaks Etcher

larrydewey, 2016-08-26:
- **Etcher version:** 1.0.0-beta.13
- **Operating system and architecture:** Arch Linux - Intel x86_64
- **Do you see any meaningful error information on DevTools?**
Nothing useful was shown in the DevTools Menu.

To reproduce, I had this file that I was writing:
`TrueOS-Desktop-11.0-CURRENTJULY2016-07-26-2016-x64-USB (1).img`
When clicking "Flash" the state freezes at "Starting", and nothing happens. A temporary work around is to remove the space and the (1) from the filename, but it is still a bug that needs to be addressed.

jviotti, 2016-08-28:
Hi @larrydewey ,

Thanks a lot for reporting this. We've fixed a very similar issue in the past, so we might be facing a small regression. Let me investigate further and report back.

jviotti, 2016-08-29:
Hi @larrydewey ,

I'm unable to reproduce on an Ubuntu 16.04 machine, which makes me wonder if there is something system specific causing this for you.

Are you running Etcher directly as `root`? If not, when the elevation dialog appears (based on `pkexec` or `kdesudo` depending in your system), you can see the actual full command being ran in the "details" section of the dialog. Can you share that here, please?

Alternatively, could it be that your file name doesn't contain a space, but another "invisible" unicode character or something like that?

cra0zy, 2016-09-08:
I am also getting the same issue. I have the same specs as @larrydewey

imrehg, 2016-09-23:
Tried this. beta14 still stalls, but the `git head` writes the same file correctly. I presume it was fixed in the changes since the report, and should be part of the next release.

jviotti, 2016-09-23:
Hm, that's weird. I'm beginning to think the issue may only be
reproducible when running inside an AppImage. Are you running with `npm
start`?

You can generate an AppImage out of `master` by running the following
command in the root of the repository:

```sh
./scripts/build/linux.sh all x64
```

The generated AppImage will be placed in `etcher-release/installers`.

Can you check if you can reproduce this way?

imrehg, 2016-09-24:
@jviotti tried it out, indeed `npm start` burns okay, `AppImage` stalls on the same image file with spaces in the name.

jviotti, 2016-09-26:
Thanks a lot for the testing @imrehg, I'll try to reproduce here and fix soon!

jviotti, 2016-09-26:
After the comment above, I'm pretty sure this issue and [this one](https://github.com/resin-io/etcher/issues/631) refer to the same problem.

closed 2016-09-26 by jviotti