balena-io/etcher #668: When run on Jessie you get an error window but after you press ok Etcher runs

Ntemis, 2016-08-29:
- **Etcher version:**
Beta 13
- **Operating system and architecture:**
Linux solydxk 3.16.0-4-amd64 #1 SMP Debian 3.16.7-ckt25-2+deb8u3 (2016-07-02) x86_64 GNU/Linux
This is what it reports as error:
Error: /tmp/.org.chromium.Chromium.fvo3eI: line 12: blkid: command not found
/tmp/.org.chromium.Chromium.fvo3eI: line 12: blkid: command not found
/tmp/.org.chromium.Chromium.fvo3eI: line 12: blkid: command not found
/tmp/.org.chromium.Chromium.fvo3eI: line 12: blkid: command not found
/tmp/.org.chromium.Chromium.fvo3eI: line 12: blkid: command not found
/tmp/.org.chromium.Chromium.fvo3eI: line 12: blkid: command not found

    at /tmp/.mount_o4fOjH/usr/bin/resources/app.asar/node_modules/drivelist/build/scripts.js:27:23
    at ChildProcess.exithandler (child_process.js:205:7)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at maybeClose (internal/child_process.js:850:16)
    at Socket.<anonymous> (internal/child_process.js:323:11)
    at emitOne (events.js:96:13)
    at Socket.emit (events.js:188:7)
    at Pipe._handle.close [as _onclose] (net.js:492:12)

jviotti, 2016-08-29:
Hi @Ntemis ,

This error was reported and resolved in https://github.com/resin-io/etcher/issues/640. The new Etcher's version will search for `blkid` on its standard location, but as a temporary fix, make sure `/sbin` is in your path.

closed 2016-08-29 by jviotti

Ntemis, 2016-08-29:
Gracias.
Great app btw keep up the good work :)

jviotti, 2016-08-29:
De nada! Let us know if you encounter any other issue!