balena-io/etcher #671: Drives without a letter assigned displayed as a blank space (followed by a separator)

WasabiFan, 2016-09-01:
- **Etcher version:** Beta 13
- **Operating system and architecture:** Windows 10
- **Do you see any meaningful error information on DevTools?** N/A

When I select a drive that doesn't have a letter assigned (either there is no partition on the drive or there isn't a partition with a letter) the drive is shown as ` - 2.0 GB`. The name is empty, so there's just a dash followed by the size.

[image]

I think it should say one of the following:
- `<no name> - 2.0 GB`
- `<unnamed> - 2.0 GB`
- `<unformatted> - 2.0 GB`
- `2.0 GB`
- `\\.\PHYSICALDRIVE3 - 2.0 GB`

The last one would be preferred, because it is the most informative.

jviotti, 2016-09-01:
Hi @WasabiFan ,

Thanks a lot for reporting this. Great catch, I didn't consider a drive without a letter assigned. We should definitely fallback to `\\.\PHYSICALDRIVEX`. I'll send a PR soon.

WasabiFan, 2016-09-01:
Now that I think about it, the physical drive information would be useful even if it has a letter (command-lime tools, logs, etc.); how easy would it be to add a tooltip or a "More info" button which lets you see that info? Obviously you can always just look at disk manager to correlate letters with physical drives, but having that info present in the UI would be nice.

jviotti, 2016-09-02:
Hi @WasabiFan ,

A tooltip or something like that should be a simple addition. Let's loop @taahirisaacs , Etcher's designer, here for his UX input first.

jviotti, 2016-09-02:
I've fixed this issue by printing the device ID if a drive letter is not assigned to the drive. Lets keep this open for tooltip/more info discussion.

jviotti, 2016-09-07:
After some discussion, it is unlikely that we will show the physical drive internal path, since its irrelevant to the big majority of our users. We will of course do show it if there is no drive letter to supersede it.

Thanks a lot!

closed 2016-09-07 by jviotti