balena-io/etcher #672: Uncaught TypeError: modal.dismiss is not a function

WasabiFan, 2016-09-01:
- **Etcher version:** Beta 13
- **Operating system and architecture:** Windows 10 x64
- **Do you see any meaningful error information on DevTools?** Yes; see below.

After Etcher finished flashing my drive, it printed an error to the console. The error doesn't seem to have affected operation in any way.

[image]

jviotti, 2016-09-01:
Hi @WasabiFan ,

Thanks a lot for reporting this. Luckily, this issue has been fixed recently by https://github.com/resin-io/etcher/pull/645, and should land in the next release!

closed 2016-09-01 by jviotti

WasabiFan, 2016-09-01:
Oh, great! GitHub's search is pretty unforgiving and apparently I didn't use it quite right.