balena-io/etcher #673: Add 0's at the beginning of Beagle images when using BMAPs

jadonk, 2016-09-01:
- **Etcher version: master**
- **Operating system and architecture: all**
- **Do you see any meaningful error information on DevTools? N/A**

There are sometimes issues with the ROM bootloader on Beagle boards when you don't 0 out the first 1MB of the SD cards before writing the images using BMAPs. This is because the bootloader expects this and the BMAP tool sees this as sparse space. If you just write 0s to the first 1MB before writing the image, it'll boot fine every time.

alexandrosm, 2016-09-01:
@jviotti how about adding an offset config parameter?

jviotti, 2016-09-02:
@jadonk What is your current workaround? Do users flashing the image with `bmaptool` frequently encounter this issue? If the zeroes are explicitly being written in the first megabyte then `bmaptool` shouldn't interpret this as sparse space. 

This sounds like the something that should be fixed from the image generation side, since it'd be a special code path in Etcher, but I'm open to further discussion.

@alexandrosm You mean an offset option to determine how much to zero out?

alexandrosm, 2016-09-03:
@jviotti exactly!

jadonk, 2016-09-03:
It could be fixed as a bug in bmaptool, good point. The writing of zeros before writing with bmaptool is the work-around. I'll raise a bug there.

RobertCNelson, 2016-09-03:
@jviotti the current workaround was to stop publishing "*.bmap" files and just recommended users use etcher.io. ;)  I'll need to retest, as it was a few quarters ago, but the gist was:

On boards like the omap3-xm (dm3730), which don't officially support "raw boot", the bootloader was found in the first fat partition.  If i had a microSD card, that was flashed for a bbb (am335x which support raw boot, (the bootrom was 2 generations later then the dm3730's..)).  So if i then used bmaptool to then flash that same card for the omap3-xm, it would fail to boot.  However if i dd'ed the image directly to the microSD, it would boot fine.  At the time since Etcher.io just came out, i then recommended users use that instead.

My theory at the time, the dm3730 was tripping over something in the 'raw boot' sectors on the microSD.  Thus clearing out the first 1MB should fix that.

Regards,

alexandrosm, 2016-09-04:
I think the solution is actually to insert the zeroes into your image and
mark them in the bmap file as an area that has to be written. This can
probably be done with manual editing of the bmap file. However the easiest
would be if @jviotti makes a configurable "pre-padding offset" and solves
it for all of us, so you can have your etcher and bmaps too :P

jviotti, 2016-09-06:
Sounds good. I'll implement and keep you updated!

jviotti, 2016-09-07:
I've started work on this today. The plan is:

- Add an option to control the amount of bytes that need to be zeroed out in https://github.com/resin-io-modules/bmapflash, our NodeJS bmap implementation module.

- Extend the `manifest.json` to support a property to configure such option in https://github.com/resin-io-modules/etcher-image-stream.

- Tied them up in Etcher itself.

jviotti, 2016-09-07:
This feature is complete, and will be merged very soon. The `manifest.json` file in an archive image may now include a `bytesToZeroOutFromTheBeginning` number option that as the name implies, signals the amount of bytes that should be zeroed out from the beginning of the drive *before* starting the bmap flash process. Setting that to 1 MB (expressed in bytes), should be enough for you!

closed 2016-09-07 by jviotti