balena-io/etcher #684: 'Starting'

asklepian, 2016-09-05:
- **Etcher-linux-x64-13.AppImage**
- **Ubuntu Mate 16.04.1 LTS**
- **Do you see any meaningful error information on DevTools?** none that I could tell...

Etcher just says 'starting' at the 3rd and final step and never changes. It doesn't do anything. I can't get it to work at all...

jviotti, 2016-09-06:
This is a duplicate of one of these issues:

- https://github.com/resin-io/etcher/issues/665
- https://github.com/resin-io/etcher/issues/631

For the first one, does your image contains parenthesis, spaces, or other weird characters? Is the error gone if you remove them?

jviotti, 2016-09-07:
I'll close this in order to continue discussion in the corresponding original issues. Please re-open if you find out neither of the issues linked above is relevant to the problem. Thanks a lot!

closed 2016-09-07 by jviotti