balena-io/etcher #685: Flashing WoT Image Fails with 'Not Enough Space'

C-D-Lewis, 2016-09-05:
- **Etcher version:** 1.0.0-beta.13
- **Operating system and architecture:** Mac OSX 10.11.6
- **Do you see any meaningful error information on DevTools?** No

Trying to flash the WoT book image from EVRYTHNG (http://webofthings.org/wot-book-pi-image/) failed with an 'not enough space' error. The SD card was a SanDisk micro SD card 16GB in size. The file is 15.9GB but has worked previously.

[image]

jviotti, 2016-09-06:
Hi @C-D-Lewis ,

Thanks for reporting this. I believe you're facing https://github.com/resin-io-modules/etcher-image-write/pull/53, which has been fixed in `master`. Can you check this custom build from `master` and see if the issue is indeed gone there? https://www.dropbox.com/s/0bqn2bq2zlu3c69/Etcher-darwin-x64.dmg?dl=0

C-D-Lewis, 2016-09-07:
Tried the linked build, but the same thing happened. This time I had console logs on hand, but doesn't seem to yield anything interesting:

```
... Flash error ...
... Drive ran out of space ...
```

jviotti, 2016-09-07:
Hi @C-D-Lewis ,

That is weird. I wonder why you get `1537 console messages are not shown`. Can you reveal them somehow (I believe you can do so using the filters menu from the toolbar)?

Can you share with me the exact size, in bytes, of both the image and the SDCard? For the image, you can do `stat -f '%z' path/to/image`. For the drive, you can do `diskutil info /dev/diskN` (where N corresponds to the SDCard), which will show something like `Total Size:               7.9 GB (7948206080 Bytes) (exactly 15523840 512-Byte-Units)`, which contains the exact bytes number (7948206080 in my case).

I believe I have an 16GB SDCard somewhere, so will try to reproduce from my side in the mid-time.

Thanks a lot!

C-D-Lewis, 2016-09-07:
Thanks for the terminal guidance. Shame each attempt takes 23 minutes!

Some numbers for you:
```
stat -f '%z' ~/Downloads/wot-pi.img
15931539456
```
diskutil info:
```
Total Size: 15.9 GB (15931539456 Bytes) (exactly 31116288 512-Byte-Units)
```

I just ran the same setup again, and did not get the warning about hidden messages. It may have been referring to the collapsing that occurs when multiple log entries contain the same content and are indicated by the blue number to the left:

[image]

jviotti, 2016-09-07:
I'm sadly unable to reproduce this issue. I've tried flashing your image, and also created a dummy image that is exactly the same size as my drive, as you've shown, but it all works fine. The only possibilities I can think of are:

- Your SD Card has a weird issue where it reports a certain size, but the real size varies a bit.
- Probably not it, but are you really sure you're running the custom build?

Do you have a smaller drive at hand? You could create a "mock" image that is exactly the size as your other drive by grabbing some data from `/dev/random` and see how that goes.

For example, say your drive is 7948206080 bytes, you can do:

```
$ dd if=/dev/random of=testing.img bs=1 count=7948206080
```

While clear, setting the block size to 1 will probably be painfully slow. You could try going for something like 65536 and dividing the count accordingly:

```
$ dd if=/dev/random of=testing.img bs=65536 count=121280
```

(7948206080 / 65536 = 121280)

You could also try building a mock image which size is equal to your drive size and check that as well.

C-D-Lewis, 2016-09-08:
I am very sure I'm using the master build, since I deleted the existing app from Applications and installed from the dmg on Dropbox. I'll give the mock image creation a go now.

C-D-Lewis, 2016-09-08:
Unfortunately I get the same result with the `testing.img` method.

jviotti, 2016-09-08:
I believe I know what the issue is. Etcher's block size defaults to 1MB. The image you have is not cleanly divisible by 1MB (it equals 15193.5 MB), which causes Etcher to pad the last block with an extra 512 kb, causing an ENOSPC error.

Most images are divisible by 1 MB, and I'm pretty sure very well-formed image will be divisible by 512 kb, so I'll add code to not pad the last block if its divisible by 512 kb and padding causes the size to exceed the drive size.

jviotti, 2016-09-08:
Hi @C-D-Lewis ,

I've PRed a fix a created a custom build for it: https://www.dropbox.com/s/sp1xi2r4ci20cvn/Etcher-darwin-x64-8-Sep-2016.dmg?dl=0.

Please let me know if this time it works for you, and I'll merge it!

C-D-Lewis, 2016-09-09:
Thanks for your continued time on this, but sadly the above build produced the same result. I'm fairly certain the image does not need to be the exact size of the card, so will try and see if it can be shrunk to size of its file contents.

jviotti, 2016-09-09:
@C-D-Lewis I honestly have no idea what could be going on in that case :/ Thank you very much for following up on this one. I'm sure we'll get to the bottom of it.

https://github.com/resin-io-modules/etcher-image-write is a re-usable module that performs writing to devices. You can think of it as Etcher's "backend". I pushed a branch there that contains a lot of debugging information that should reveal what the problem is. It'd be amazing if you can give it a quick go and share the output back in this thread.

You can get it like this:

- Make sure you have NodeJS installed. Any modern version (because of ES6) works. I'm on v6.1.0 here.
- `git clone https://github.com/resin-io-modules/etcher-image-write`
- `cd etcher-image-write`
- `git checkout c-d-lewis-debug-685`
- `npm install`

Then you can run the `example.js` file like this:

```
sudo node example.js path/to/image.img /dev/diskN
```

Let me know how it goes, and thank you very much again for the kind help!

C-D-Lewis, 2016-09-09:
OK, I followed those (detailed!) instructions and have something that may be of interest:

```
-----------------------------------------------
Drive size: 15931539456
Image size: 15931539456
Block size: 1048576
Chunk length: 1048576
Transferred bytes so far: 15928918016
Transferred bytes after chunk: 15929966592
Chunk pushed!
-----------------------------------------------
Drive size: 15931539456
Image size: 15931539456
Block size: 1048576
Chunk length: 1048576
Transferred bytes so far: 15929966592
Transferred bytes after chunk: 15931015168
Chunk pushed!
-----------------------------------------------
Drive size: 15931539456
Image size: 15931539456
Block size: 1048576
Chunk length: 1048576
Transferred bytes so far: 15931015168
Transferred bytes after chunk: 15932063744
Not enough space on the drive
```

It does appear to overstep the capacity in the last loop before failure, if I'm reading it correctly?

jviotti, 2016-09-09:
@C-D-Lewis Thank you very much for the quick response. So this shows us I was right in my initial assumption about the final chunk being padded to reach 1 MB. However this means my fix should have worked.

I've rebased the branch I sent you above to include my fix. Do you mind re-running with those changes, and see if find out something weird on my latest fix? I force pushed to the branch, so you'll need to either force pull, delete the branch and re-fetch, or clone again!

Thank you very much!

C-D-Lewis, 2016-09-12:
Looks like this newest version solves the problem! I see a non-error result at the end:

```
-----------------------------------------------
Chunk length before padding: 524288
Chunk length after padding: 524288
Drive size: 15931539456
Image size: 15931539456
Block size: 1048576
Minimum block size: 524288
Transferred bytes so far: 15931015168
Transferred bytes after chunk: 15931539456
Chunk pushed!
{ sourceChecksum: 'c316afa4' }
```

and the image boots successfully! Thanks for your help on this, and let me know when the fix makes its way into a release.

jviotti, 2016-09-12:
Hi @C-D-Lewis ,

I'm really happy this worked for you, however what you're running in `etcher-image-write` is the exact same code as the latest custom Etcher built I sent you, so it should have worked there.

Any ideas? Did something change between your tests (actual drive, etc)?

C-D-Lewis, 2016-09-14:
As far as I know, nothing changed. The image was left as unpacked, and the SD card was the same.

jviotti, 2016-11-03:
I revised my previous fix after other fixes took place in the writer, and I can confirm the issue completely gone for me. This fix will be part of the next Etcher version. Please re-open if you still experience problems!

closed 2016-11-07 by jviotti