balena-io/etcher #697: Add support for post-flash image instructions

jviotti, 2016-09-12:
Publishers will be able to include a markdown file in an image archive, that we will show (if it exists) to the user once the flash completed, as a way for publishers to provide sensible "next steps".

Some thoughts:

- The instructions should be shown by default. Of course, the instructions document will probably be too big to be shown at once on the screen for most images, so we could inline it on the completion screen and add a scroll bar or something like that.

- This feature means that the completion screen will have to be re-designed to make some space for the instructions. It also means that even if the image doesn't ship with instructions, the whole thing should look good as well.

jviotti, 2016-09-12:
We have a prototype for this feature here already https://github.com/resin-io/etcher/pull/694

taahirisaacs, 2016-09-15:
Ola @jviotti I've created a design with two layouts (right and left) 

[image]
[image]

taahirisaacs, 2016-09-15:
@jviotti Updated version!

[image]