balena-io/etcher #700: Error Starting Flash with beta 14

wsloand, 2016-09-13:
- **Etcher version:** 1.0-beta14
- **Operating system and architecture:** Windows 10 x64
- **Do you see any meaningful error information on DevTools?** 

tracker.js:10 Error: { Error: Cannot find module 'elevator'
    at Function.Module._resolveFilename (module.js:438:15)
    at Function.Module._load (module.js:386:25)
    at Module.require (module.js:466:17)
    at require (internal/module.js:20:19)
    at isElevated.then (C:\Program Files (x86)\Etcher\resources\app.asar\lib\src\child-writer\writer-proxy.js:84:46)
    at tryCatcher (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\bluebird\js\release\util.js:16:23)
    at Promise._settlePromiseFromHandler (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\bluebird\js\release\promise.js:504:31)
    at Promise._settlePromise (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\bluebird\js\release\promise.js:561:18)
    at Promise._settlePromise0 (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\bluebird\js\release\promise.js:606:10)
    at Promise._settlePromises (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\bluebird\js\release\promise.js:685:18)
    at Promise._fulfill (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\bluebird\js\release\promise.js:630:18)
    at C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\bluebird\js\release\nodeback.js:42:21
    at C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\is-admin\index.js:12:3
    at ChildProcess.exithandler (child_process.js:221:5)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7) code: 'MODULE_NOT_FOUND' }

    at Socket.child.stderr.on (C:\Program Files (x86)\Etcher\resources\app.asar\lib\src\child-writer\index.js:83:29)
    at emitOne (events.js:96:13)
    at Socket.emit (events.js:188:7)
    at readableAddChunk (_stream_readable.js:172:18)
    at Socket.Readable.push (_stream_readable.js:130:10)
    at Pipe.onread (net.js:542:20)
[image]

curcuz, 2016-09-13:
Hi, similar issue - same etcher version, same OS, different module

```
ue Sep 13 2016 11:41:38 GMT+0200 (W. Europe Daylight Time) Application start
tracker.js:10 Tue Sep 13 2016 11:41:38 GMT+0200 (W. Europe Daylight Time) Checking for updates
tracker.js:10 Tue Sep 13 2016 11:41:43 GMT+0200 (W. Europe Daylight Time) Select image ({"size":1526726656,"path":"C:\\Users\\czkzc\\Downloads\\resin-boombeasticrpi3-1.8.0-1.13.0-7ad55d1c946b.img","logo":false,"bmap":false})
tracker.js:10 Tue Sep 13 2016 11:41:45 GMT+0200 (W. Europe Daylight Time) Flash ({"image":"C:\\Users\\czkzc\\Downloads\\resin-boombeasticrpi3-1.8.0-1.13.0-7ad55d1c946b.img","device":"\\\\.\\PHYSICALDRIVE1"})
tracker.js:10 WRITER: Tailing C:\Users\czkzc\AppData\Local\Temp\etcher-6680kl8z6lwhiHT3.log

tracker.js:10 WRITER: Re-spawning with elevation

tracker.js:10 Error: module.js:440
    throw err;
    ^

Error: Cannot find module 'diskpart'
    at Function.Module._resolveFilename (module.js:438:15)
    at Function.Module._load (module.js:386:25)
    at Module.require (module.js:466:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\etcher-image-write\lib\win32.js:26:39)
    at Module._compile (module.js:541:32)
    at Object.Module._extensions..js (module.js:550:10)
    at Module.load (module.js:456:32)
    at tryModuleLoad (module.js:415:12)
    at Function.Module._load (module.js:407:3)
    at Module.require (module.js:466:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\etcher-image-write\lib\index.js:25:13)
    at Module._compile (module.js:541:32)
    at Object.Module._extensions..js (module.js:550:10)
    at Module.load (module.js:456:32)
    at tryModuleLoad (module.js:415:12)
    at Function.Module._load (module.js:407:3)
    at Module.require (module.js:466:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (C:\Program Files (x86)\Etcher\resources\app.asar\lib\cli\writer.js:19:20)
    at Module._compile (module.js:541:32)
    at Object.Module._extensions..js (module.js:550:10)
    at Module.load (module.js:456:32)
    at tryModuleLoad (module.js:415:12)
    at Function.Module._load (module.js:407:3)
    at Module.require (module.js:466:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (C:\Program Files (x86)\Etcher\resources\app.asar\lib\cli\etcher.js:24:16)
    at Module._compile (module.js:541:32)
    at Object.Module._extensions..js (module.js:550:10)
    at Module.load (module.js:456:32)
    at tryModuleLoad (module.js:415:12)
    at Function.Module._load (module.js:407:3)

    at Socket.child.stderr.on (C:\Program Files (x86)\Etcher\resources\app.asar\lib\src\child-writer\index.js:83:29)
    at emitOne (events.js:96:13)
    at Socket.emit (events.js:188:7)
    at readableAddChunk (_stream_readable.js:172:18)
    at Socket.Readable.push (_stream_readable.js:130:10)
    at Pipe.onread (net.js:542:20)a.(anonymous function) @ tracker.js:10
tracker.js:10 Tue Sep 13 2016 11:42:22 GMT+0200 (W. Europe Daylight Time) Flash error
tracker.js:10 Tue Sep 13 2016 11:42:26 GMT+0200 (W. Europe Daylight Time) Restart after failure
```

jviotti, 2016-09-13:
Hi @wsloand , @curcuz ,

Turns out we've been hitting an npm bug on NodeJS v4.5.0 on our Windows build machine that incorrectly omits all optional dependencies.

I'm publishing a new version now, for which I'll ping you once its uploaded.

Sorry for the inconveniences!

jviotti, 2016-09-13:
Done! Re-downloading v1.0.0-beta.14 from the website should fix the issue!

closed 2016-09-13 by jviotti

wsloand, 2016-09-14:
The new download fixed that issue, but I've got a new one.  I'll open a new issue.