balena-io/etcher #701: Add ability to create an image from a USB/SD card for backups/etc

ricecooker, 2016-09-13:
- **Etcher version:** 1.0.0-beta.13
- **Operating system and architecture:** OSX 10.11.6
- **Do you see any meaningful error information on DevTools?** No, feature request

Add a feature to write an image to disk as an ```.img``` or ```.iso``` to make backups and restores easier.  After burning an image, the first thing people do is customize it for their use.  Being able to easily backup it up to use on another sd card for another sensor would be great.

dlech, 2016-09-13:
Duplicate of #266

jviotti, 2016-09-13:
Hi @ricecooker ,

This is feature we're looking to implement very soon, although we don't have an estimate for it at the moment!

Thanks a lot for the heads up @dlech ! I'll close this one and we can continue discussion in the original one!

closed 2016-09-13 by jviotti

ricecooker, 2016-09-13:
Sorry about the dupe.  Thanks all.

rin67630, 2019-08-13:
An update on this? Has someone got a procedure to otherwise create Balena compatible images?
I have tested dd but with that image Balena got stuck and wrote nothing to the SD card.