balena-io/etcher #702: Error: EPERM: operation not permitted, write

wsloand, 2016-09-14:
- **Etcher version:** 1.0-beta14
- **Operating system and architecture:** Windows 10 x64
- **Do you see any meaningful error information on DevTools?** 

After this, I ran diskpart (select the disk) clean, and I still got the error.

Wed Sep 14 2016 14:42:50 GMT-0400 (Eastern Daylight Time) Progress (write): 1% at 6.38 MB/s (eta 436s)
tracker.js:10 Error: EPERM: operation not permitted, write
    at ChildProcess.child.on (C:\Program Files (x86)\Etcher\resources\app.asar\lib\src\child-writer\index.js:92:23)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at handleMessage (internal/child_process.js:718:10)
    at Pipe.channel.onread (internal/child_process.js:444:11)a.(anonymous function) @ tracker.js:10
tracker.js:10 Wed Sep 14 2016 14:44:51 GMT-0400 (Eastern Daylight Time) Flash error

[image]

wsloand, 2016-09-14:
An additional note:  I downloaded beta-13, and it appears to be working (it at least got much farther-- the beta-14 attempt ended around 1% written).

jviotti, 2016-09-14:
Hi @wsloand ,

Thanks a lot for reporting this. These `EPERM` Windows issues are a nightmare. There is no way to know the reason they are thrown, and they seem to happen in random ways.

Can you link me to the image you're trying to flash?

Some thoughts:

- What happens if you try to flash other images to that exact same drive? Does it fail with EPERM as well?
- What happens if you try to flash the same image to other drives?
- What happens if you clean the drive with `diskpart.exe`, like you did, but trigger a `rescan` command before running Etcher?
- Does the error happens every single time, or does it work sometimes if you repeatedly retry?

There is another GitHub issue about this exact same problem (https://github.com/resin-io/etcher/issues/627), so probably worth continuing discussion there, so we keep all information in a single place.

I'll close this one!

closed 2016-09-14 by jviotti