balena-io/etcher #708: zenity errors from AppImage

glensc, 2016-09-20:
- **Etcher version:** 1.0.0-beta14
- **Operating system and architecture:** PLD Linux x86_64

AppImage launcher spills some notices to terminal. otherwise it launches fine, but likely some zenity message i failed to see

```
[~/Downloads] ➔ ./Etcher-linux-x64.AppImage 
/tmp/.mount_lynvJD/usr/bin/etcher.wrapper

(zenity:17783): Gtk-WARNING **: Could not find the icon 'dialog-question-ltr'. The 'hicolor' theme
was not found either, perhaps you need to install it.
You can get a copy from:
	http://icon-theme.freedesktop.org/releases

(zenity:17783): Gtk-WARNING **: Could not load a pixbuf from /org/gtk/libgtk/icons/48x48/status/image-missing.png.
This may indicate that pixbuf loaders or the mime database could not be found.
**
Gtk:ERROR:gtkiconhelper.c:493:ensure_surface_for_gicon: assertion failed: (destination)
./bin//etcher.wrapper: line 68: 17783 Aborted                 (core dumped) LD_LIBRARY_PATH="" zenity --question --title="$TITLE" --text="$TEXT"
```

jviotti, 2016-09-20:
Hi @glensc ,

The issue comes from the `desktopintegration` AppImage script, which attempts to show a dialog to create a link to the AppImage in your applications menu.

As a quick workaround, you can disable this AppImage feature using the following instructions: https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#deactivate-desktop-shortcut-prompt-on-gnulinux

For a real fix, I suggest re-posting this in the AppImage project (https://github.com/probonopd/AppImageKit) and see what they think. If they have a fix, happy to upgrade the AppImages tooling here.

jviotti, 2016-09-21:
I'm closing this issue since it looks like a GTK bug (see https://bugzilla.gnome.org/show_bug.cgi?id=765649), or a GTK configuration issue. What GTK version are you running?

Please re-open if you find any insights.

closed 2016-09-21 by jviotti

glensc, 2016-09-21:
gtk+2-2.24.30-1.x86_64 / gtk+3-3.20.6-1.x86_64 (not sure is gtk2 or gtk3 relevant)