balena-io/etcher #710: Validation error for some bz2 compressed image

imrehg, 2016-09-21:
- **Etcher version:** starting from commit a4e3fcccbacf93010f5df21dbf7cbc8cb755b507
- **Operating system and architecture:** ArchLinux x64
- **Do you see any meaningful error information on DevTools?** none, just validation error at the end of process

The commit linked above (upgrade: `etcher-image-stream` to v4.3.0 (#699)) fixed some other buggy write, but starting from that commit, writing some `bz2` image results in validation error. Same image uncompressed, or compressed with `xz` or `zip` flashes fine (`gz` has issue but likely separate, mentioning it not to confuse).

A known failing file: https://www.dropbox.com/s/wwnv8whi9b7xbs4/resin-hostname-1.8.0-1.13.0-7ad55d1c946b.img.bz2?dl=0

jviotti, 2016-09-21:
Hi @imrehg ,

Thank you for the valuable information! Can you try if decompressing with https://github.com/regular/unbzip2-stream directly vs decompressing with the `bzip2` tool both lead the exact same data (with matching checksums)?

This might be an issue with the bzip2 decompression algorithm implemented in the module we're using. If this is the case, lets move this issue to that repository (I don't have the knowledge to debug issues in the bzip2 decompressor implementation).

jviotti, 2016-09-21:
However, even if https://github.com/regular/unbzip2-stream decompressed incorrectly, the checksum between what was retrieved from the archive and what was written to the device should have matched :/

imrehg, 2016-09-21:
@jviotti tested and the checksum matches for the two case.

jviotti, 2016-09-21:
I can indeed reproduce here. I'll investigate and report back very soon.

jviotti, 2016-09-21:
I've found the issue. On the validation phase, we only consume bytes up to a certain point from the drive (up to the input image size), however for bz2 in particular, we're using the "compressed" image size to slice the stream, causing Etcher to be comparing a subset of the image to the whole image (which of course fails).

I'll send a fix very soon.

jviotti, 2016-09-21:
Here is a custom build with the change: https://dl.dropboxusercontent.com/u/18129309/Etcher-linux-x64.AppImage

Can you confirm it solves both gz and bzip2 issues for you?

closed 2016-09-22 by jviotti

imrehg, 2016-09-23:
Sorry for the delay, yes, seems to write well on all formats :)