balena-io/etcher #716: Suggestion for displaying image metadata (logo and description)

dlech, 2016-09-23:
Currently, when you load an image with metadata, it will look something like this:

[image]

I would like to suggest that instead of replacing the filename with the "name" from the metadata, that "SELECT IMAGE" is replaced instead to show that the image has been selected. It would look something like this.

[image]

BertLindeman, 2016-09-23:
Maybe not really at it's place here, but could there be a little better contrast between the gray background and the gray text? Or maybe a slightly larger font size? I have to take a photo to be able to read the text. As an example in the image above the text "show full file name".

jviotti, 2016-09-26:
Hi @dlech ,

Thanks a lot for the suggestion. I definitely agree with it, and will make the changes very soon.

@BertLindeman 

Thanks for raising this up. What do you think @taahirisaacs ?

lurch, 2016-09-30:
I agree that small grey text on a grey background is not the easiest thing to read (I imagine it's even worse on a screen with reduced colour depth!)

Also: would it be possible to display the full-image-path (e.g. as a tooltip) when mouse-hovering over the "SHOW FULL FILE NAME" ?

WasabiFan, 2016-09-30:
> Also: would it be possible to display the full-image-path (e.g. as a tooltip) when mouse-hovering over the "SHOW FULL FILE NAME" ?

See https://github.com/resin-io/etcher/issues/634#issuecomment-240274982

jviotti, 2016-09-30:
I guess we could, but is much harder to protect from long file names causing the tooltip from overflowing the application screen. For simplicity purposes, I think the modal is the way to go, but maybe we can show something else as the tooltip, like the image basename, or truncate the absolute file path, although that might not be very useful anyway.

lurch, 2016-09-30:
> but is much harder to protect from long file names causing the tooltip from overflowing the application screen

Ahh, I hadn't considered that. In which case it probably makes sense to keep the "SHOW FULL FILE NAME" button as a modal, and instead show the full basename of the image when mouse-hovering over the truncated image name ("nightly-ev3d...09-04.img.xz" in the screenshot above).
And perhaps you could also show the full drive-name in a tooltip when mouse-hovering over the drive-name too, in case that ever gets truncated (unlikely, but you never know!).

jviotti, 2016-09-30:
Completely agree with the suggestions. Would you like to take a shot at it? Adding a tooltip is just a matter of adding a `uib-tooltip` attribute to the elements. For example:

```html
<p class="step-footer">
  {{ ::image.mainSupportedExtensions.join(', ') }}, and
  <span class="step-footer-underline"
    uib-tooltip="{{ image.extraSupportedExtensions.join(', ') }}">many more</span>
</p>
```

And you can get the base name of a path using the `path.basename` function (https://nodejs.org/api/path.html#path_path_basename_path_ext).

lurch, 2016-09-30:
> Would you like to take a shot at it?

CHALLENGE ACCEPTED: #736 
;-)

I **think** the image-name stuff is right, but I wasn't able to test it, because despite struggling to follow the metadata-instructions in #707 and create an 'extended zipfile', I eventually found that the code is actually doing something different :-(
(#707 says `.meta/manifest.json` but the code is actually looking for `_info/manifest.json`)

lurch, 2016-09-30:
It would also be nice if the image filename didn't appear as clickable-on (i.e. without the hand-cursor when hovering over it) if the image _doesn't_ have a URL set, but I couldn't tell how to only add the `ng-click` attribute conditionally if imageURL has been set.

jviotti, 2016-10-03:
> I think the image-name stuff is right, but I wasn't able to test it, because despite struggling to follow the metadata-instructions in #707 and create an 'extended zipfile', I eventually found that the code is actually doing something different :-(
(#707 says .meta/manifest.json but the code is actually looking for _info/manifest.json)

Yeah, it was `_info` originally, but I believe `master` already reads `.meta`. Isn't that the case?

> It would also be nice if the image filename didn't appear as clickable-on (i.e. without the hand-cursor when hovering over it) if the image doesn't have a URL set, but I couldn't tell how to only add the ng-click attribute conditionally if imageURL has been set.

You could accomplish that by using [`ngStyle`](https://docs.angularjs.org/api/ng/directive/ngStyle) and [`cursor: default`](It would also be nice if the image filename didn't appear as clickable-on (i.e. without the hand-cursor when hovering over it) if the image doesn't have a URL set, but I couldn't tell how to only add the ng-click attribute conditionally if imageURL has been set.)

dlech, 2016-10-03:
A search for `_info` and `.meta` both come up empty. Which file would it be in?

dlech, 2016-10-03:
Given the fact that beta15 works with archives that use `_info`, I'm guessing `master` doesn't use `.meta`.

jviotti, 2016-10-03:
Indeed, my bad! The change landed in [etcher-image-stream](https://github.com/resin-io-modules/etcher-image-stream/blob/master/CHANGELOG.md#v500---2016-09-14), but the dependency hasn't been upgraded yet.

jviotti, 2017-01-24:
I'm closing this given the new approach we're going to take towards image metadata.

closed 2017-01-24 by jviotti