balena-io/etcher #717: Update drive icon/text similar to the way image icon/text are update when metadata is present

dlech, 2016-09-23:
This idea follows on #716. The idea is for the drive icon and text to be updated when a drive is selected to keep the behavior patterns of the program the same for each step of the 3 steps shown.

Since the disk image icon and text can be updated when an image is selected, it follows that the disk drive icon and text should be updated in a similar manner. Here is an example...

[image]

I don't know what kind of info is available from the OS about what type a drive is connected (e.g. USB/SD card/etc.) but it would be slick if the icon could be updated to show what type of drive was selected.

alexandrosm, 2016-09-23:
We really wanted to make the icon be updated on usb/sd drive, but it seems
there's no reliable way to tell, especially across platforms. it would even
be cool if we could do it in a subset of cases we could detect, but we've
found no way to get even partial information as of yet.

alexandrosm, 2016-09-23:
otherwise, I do agree on both the suggestions and indeed we've discussed
similar changes in the past. This is a good nudge to revise that
conversation (pinging @taahirisaacs)

dlech, 2016-09-23:
Yes, detecting this would be very platform-specific. It [looks like you are already getting this info for Linux](https://github.com/resin-io-modules/drivelist/blob/master/scripts/linux.sh#L56) but just not saving it.

dlech, 2016-09-23:
So, essentially, this issue is blocked by resin-io-modules/drivelist#64

alexandrosm, 2016-09-23:
nice! I'd be up for doing it in platforms that support it.

jviotti, 2016-09-26:
I definitely agree with this change. Maybe we can get the ball rolling for just Linux for now (seems to be the lowest hanging fruit)?

lurch, 2018-02-23:
As Alex said earlier, this info seems to be very unreliable; even on desktop Ubuntu 16.04 my SD card-reader gets shown with a CompactFlash icon :confused: